genodelabs/genode #2831: nit_fader: still shining through with alpha value of 255

BorisMulder-CSL, 2018-05-17:
When I use a nit_fader to display a backdrop that is completely black (<fill color="#000000"/>) and set the alpha to 255, I can still see the underlying layer faintly.

I assume 255 is the maximum as it is usual for alpha in the graphics and it is the maximum value in nit_fader.run (which does not work by the way, it has no cap quota defined and seems to be old). This maximum value is not defined though.

The main reason to want this is to implement a screensaver that dims the screen after some time and displays a black screen after more time.

chelmuth, 2018-05-17:
@BorisMulder-CSL a patch is much appreciated esp. for the debug run script, which is seldom used.

BorisMulder-CSL, 2018-05-17:
Fixed it

nfeske, 2018-05-21:
The artifacts are caused by the dithering. Since the dithering noise is applied to the alpha values (subtractive), the effective alpha value becomes lower than 255. You can see the corresponding code here: https://github.com/genodelabs/genode/blob/master/repos/os/include/os/dither_painter.h

The effect will disappear once we switch to a 32-bit color format that won't require dithering.

BorisMulder-CSL, 2018-05-22:
All right, thank you for your comment. Since this issue is not on short-term a pressing concern for us, we might want to wait until the "color update" will be there. At least the cause is clear to me now.

commit f7fc822a (2019-01-03): nitpicker/nit_fader: avoid color bleeding

This patch improves the output of opaque pixels in the presence of an
alpha channel by adding a special case for the maximum alpha value.

Fixes #2831
closed 2019-01-07 by nfeske