genodelabs/genode #2833: simplify fs_rom

ehmry, 2018-05-18:
Fs_rom could be simpler.

ehmry, 2018-05-18:
Patch at a0a716119f0cb0471881929bc38d1cd6d7106f8d.

nfeske, 2018-05-22:
@ehmry closely related to this topic, I got new insights into the mysterious ID-space conflict when using fs_rom in the sculpt scenario. I can reproduce the problem now and found that fs_rom is not at fault. Instead, the problem is caused by the use of fs_rom (as child service) for the environment sessions other children. Apparently, init fails to wait for the response of close requests before issuing new session requests to fs_rom. I'm at it.

ehmry, 2018-05-29:
@chelmuth I have two commits to fix the regression in `fs_rom`, 55717b7fb9d62bbd0d9c49cfaa07388e0f71e65f and d5fce652c574a311ee7d2a4dce5ec6dfd53f2d25. See my **fs_rom** branch.

The last patch to fs_rom was from older branch, and revert the component to a version without notification for empty files.

chelmuth, 2018-05-29:
Thanks, the patches fix the run/fs_report issue. Unfortunately, the run/depot_query test is still broken (since ba91a2258803f24a0ba3f536071d4e08f59321ab). Could you please have a look?

commit 00282ac0 (2018-05-22): Simplify fs_rom internals

Merge the 'Packet_handle' into the 'Rom_root' and use an Id_space to
match File_system handles to session objects rather than a list. This
avoids the need to override 'Root_component::_destroy_session'.

Fix #2833
closed 2018-05-31 by chelmuth