genodelabs/genode #2835: platform_drv: don't disable PCI bus master DMA of devices during platform_drv bootstrap

alex-ab, 2018-05-22:
Some devices are in use by the BIOS/UEFI, e.g. USB. For such devices explicit protocols (handover) exists to pass control to the operating system. By switching of PCI bus master DMA without following the protocol the whole machine my get stuck and hang hard. Observed on a AMD board.

alex-ab, 2018-06-01:
@chelmuth: based on our discussion please review 40c41b393c2210e1ab220b059df332a8bdc30749

chelmuth, 2018-06-01:
I like the commit but hesitate to merge it the next two weeks, as long as I'm not convinced that we achieve reasonable test coverage before the sculpt image release. Opinions anyone?

alex-ab, 2018-06-01:
I understand, in principle. ;-) But ...

The argument against waiting goes as follow:
- The commit fixes hard hangs during boot due to USB handover protocol violation by Genode on some (few) machines - which would be a bad Sculpt experience ...
- By adding it early on sculpt_next, we actual would get the test coverage on all of our machines.
- If it turns out to break things on some of our machines, we can drop it immediately. 

But we may wait of course, I fully understand why. I will not insists ;-)

chelmuth, 2018-06-04:
I merged 0ba5fc0b59b6c6e590b0602bfa5747eb9fff0a34 and appreciate any feedback regarding regressions.

alex-ab, 2018-06-04:
Works for me on 7 native machines (Sculpt 18.05 + this patch)

nfeske, 2018-06-04:
@alex-ab thanks for the elaborative testing. I have also switched my installation to the most current version.

commit a64fe425 (2018-05-23): platform_drv: don't disable devices during boot

Fixes #2835
commit 0ba5fc0b (2018-06-04): platform_drv: disable PCI DMA class specific

for such classes where it should be safe and where we have seen issues.
Disabling in general bus master DMA causes on some machines hard hangs, e.g.
because the USB handover protocol was violated.

Fixes #2835
commit 5ca30b03 (2018-06-12): platform_drv: disable PCI DMA class specific

for such classes where it should be safe and where we have seen issues.
Disabling in general bus master DMA causes on some machines hard hangs, e.g.
because the USB handover protocol was violated.

Fixes #2835
closed 2018-06-13 by chelmuth