genodelabs/genode #2836: usb_block_drv: make device reset optional

cproc, 2018-05-22:
The 'bulk-only mass storage reset' command, which is currently sent to the device at the beginning of the initialization step, caused write command errors on a 'SanDisk Cruzer Force' (0781:5572) USB stick. The 'usb_drv'-internal storage driver does not send the reset command and omitting it from 'usb_block_drv' didn't seem to cause any problems with the USB storage devices I had avaliable for testing, so we could omit it by default and provide a 'reset_device' config attribute for the case that a reset is needed for some device.

chelmuth, 2018-06-01:
Sculpt with packages from 18.05 does not always successfully *inspect* the boot thumb drive. I could reproduce the issue with Corsair Voyager (not GT) and SanDisk 32 GB micro drives. @cnuke suspects the commit above and also knew the workaround: Remove and replug the boot drive before inspect. I suggest the block-driver state machine should be changed to reset the device if the inquiry command fails initially. This issue is critical for the best Sculpt experience!

cproc, 2018-06-11:
Inspecting the mentioned devices should work with the commit from issue #2860.

closed 2018-06-13 by cproc