genodelabs/genode #2839: fs_rom does not notify when files are removed

ehmry, 2018-05-27:
The `fs_rom` server is certainly expected to send signals to clients when file content is replaced. It should also send signals when files are removed. This does not happen currently because `fs_rom` treats an empty file as a transition between modifications, but incorrectly treats a missing file as a zero-length file. It is important to send signals on file removal because the `fs_rom` defers moving its watch handles up and down the file-system hierarchy until a client makes a request.

Better behavior would be:
- Server discovers file is removed by a watch notification, client is signaled
- Client receives signal and requests ROM update
- Server acts on missing file by moves file watch handle to the nearest parent directory, responds to client
- Client examines dataspace and finds it empty
- ...
- Server is notified of a change to directory, signals client
- Client requests ROM update
- Server opens and reads new file and responds to client
- Client has current file content

ehmry, 2018-05-27:
@nfeske fixed at f1d50ea8b7d6a3fcf53545c757a70c415f1bf526.

ehmry, 2018-05-27:
@nfeske Fix for `Unknown_id` at a9694805339234b82ccd15860a141e5d081b27e8.

ehmry, 2018-05-30:
@chelmuth I have a fixup at 7adb1a4acb29ed9e63f4be5590745adc61082773 to throw out the signaling heuristics in fs_rom and notify the appropriate client for every file-system notification. That would ensure that fs_rom never willfully withholds a notification, and should not perform worse than the current conditions where fs_rom is making synchronous stat calls in response to signals.

If there are problems or a desire for optimization, my solution would be to add a version number field to the VFS/File-system stat as inspired by 9P (see `qid.vers` in  the [9P stat manpage](http://man.cat-v.org/plan_9/5/stat)). This would be difficult to extend from the in-memory file-systems to ext2, but perhaps not impossible.

chelmuth, 2018-05-30:
The patch prevents the test from getting stuck, but I get following error in about 20% percent of the runs.
```
make -C x86_32 run/fs_report KERNEL=pistachio
...
Genode 18.02-478-g2ee80e2
490 MiB RAM and 64534 caps assigned to init
[init -> test-fs_report] --- test-fs_report started ---
[init -> test-fs_report] (1) check initial content of "devices" ROM
[init -> test-fs_report] (2) issue new "devices" report before installing a ROM signal handler
[init -> test-fs_report] (3) wait a bit to let the report reach the RAM fs
[init -> test-fs_report] (4) install ROM signal handler, is expected to trigger immediately
[init -> test-fs_report] (5) received ROM update as expected
[init -> test-fs_report] (6) request not-yet-available "focus" ROM
[init -> fs_rom] /focus ROM file is missing
[init -> test-fs_report] (7) wait a bit until generating the focus report
[init -> test-fs_report] (8) generate "focus" report, is expected to trigger ROM notification
[init -> test-fs_report] (9) received expected focus ROM content
[init -> test-fs_report] (10) remove focus file
[init -> fs_rom] /focus ROM file is missing
[init -> fs_rom] /focus ROM file is missing
[init -> test-fs_report] Error: (9) unexpected content of "focus" ROM
Kernel: Thread 'ep' died because of an uncaught exception
[init -> test-fs_report] Error: Uncaught exception of type 'Genode::Exception'
[init -> test-fs_report] Warning: abort called - thread: ep
[init] child "test-fs_report" exited with exit value 1
```

The log shows that an unexpected signal handler is executed in the test program and that this issue is not linux-specific but also happens on pistachio (as well as sel4, okl4, and foc). The cause may be an invalid assumption of the test program itself, e.g., if a second signal was queued in the current handler before the next signal context is registered at fs_rom. Note, the handlers are not destructed and, so, may fire again.

ehmry, 2018-05-30:
I suspected there were signals queuing, but that can be fixed at fs_rom by tracking if the client has responding to a signal. Fixed at dcbb2b650af3d8429ae6d4f9d8f99000e9c84140?

closed 2018-05-31 by chelmuth

chelmuth, 2018-06-06:
@ehmry could you please look again into the (spurious signal) issue of the fs_report test? We have at 20 tests failing each night because of this.

ehmry, 2018-06-07:
I've adjusted the fs_report test at c35d2862849a49018de300dd1ab721bb220686d2, that should allow the test to handle a requed signal.

chelmuth, 2018-06-07:
I got your idea but I'm still wondering about the origin of the spurious signal because my log under okl4 is now.
```
...
[init -> test-fs_report] (8) generate "focus" report, is expected to trigger ROM notification
[init -> test-fs_report] (9) received expected focus ROM content
[init -> test-fs_report] (9) received expected focus ROM content
[init -> test-fs_report] (10) remove focus file
...
```
Mind the double `(9)`. Could you look a bit deeper into this?

ehmry, 2018-06-07:
Fixed at 8ee49b2b9fe75ed16c269bb7b9d94486c3e46adf.

commit ba91a225 (2018-05-28): Fs_rom: send signals on file removal

Clients may wish to act on missing files. In any case the fs_rom
needs to reopen a watch handle when a file is deleted, and this
sort of change to the internal state machine is propelled by
client RPC requests.

Fix #2839
commit 1af6bd14 (2018-05-31): Fs_rom: send signals on file removal

Clients may wish to act on missing files. In any case the fs_rom
needs to reopen a watch handle when a file is deleted, and this
sort of change to the internal state machine is propelled by
client RPC requests.

Fix #2839