genodelabs/genode #2840: nic_router: arbitrary uplink sessions at any domain

m-stein, 2018-05-27:
Instead of the implicitely special role of the uplink with a reserved domain name and exactly one NIC session that is bound to this domain, I want to make it possible to let the NIC router request multiple uplinks for any domain by introducing a new sub-tag to the domain tag:
```
<domain name="alpha">
   <uplink label="wifi" />
<domain />
<domain name="beta">
   <uplink label="wired_1" />
   <uplink label="wired_2" />
<domain />
```
The lifetime of these uplinks is still bound to the lifetime of the domains and they get re-requested on a change of the label attribute.

m-stein, 2018-05-27:
The uplinks would of course coexist with the downlinks of their domain in almost be treated the same (except when it comes to reading out their MAC address or link state).

m-stein, 2018-06-05:
Fixed by:
bf8e053 nic_router: multiple uplinks
729f2f0 nic_router: minor log fixes in Domain constructor
ae72fbe nic_router: get rid of the term "legacy"
7957018 nic_router: fix domain invalidation during init
1de79fe nic_router: invalidate domains with same name
741a1e2 nic_router: generic Avl_string_tree
ea31ec5 nic_router: simplify routing rule classes
0ffdd28 session_label: make typedef String public
ca4ea08 session_label: fix ineffective protection

chelmuth, 2018-06-06:
@m-stein I'd like to merge fixes for Sculpt first and delay features for the next two weeks. Could you please mark changes that are fixes? Also, did you address the following sporadic issue (I experienced the last time on Monday)?
```
[runtime -> nic_router] Warning: waiting for ARP cancelled
[runtime -> nic_router] Error: __cxa_pure_virtual called, return addr is 0x1032d5f
[runtime -> nic_router] Warning: abort called - thread: ep
```

m-stein, 2018-06-06:
I update the branch so that the fixes come first. Those are fixes:
7087d12 nic_router: fix domain invalidation during init
2b686dc nic_router: invalidate domains with same name
523537f nic_router: generic Avl_string_tree
5d23c5e nic_router: simplify routing rule classes
f63d931 ping.run: mark as "not supported" on Linux & FOC

5d23c5e is a fix because it ensures that the router does not try to apply domains with a broken configuration.

m-stein, 2018-06-06:
I'll now concentrate on the so-far reported bugs.

chelmuth, 2018-06-06:
I'm at the merge. Does your note about 5d23c5e39f74bb79d742b48d632721385bcd362c qualify as an answer to my question regarding the abort?

m-stein, 2018-06-13:
@chelmuth I've reworked the remaining commits of this issue (and added some) to - amongst other things - solve the recent problems with run/nic_router_uplinks:
bf3da4d nic_router: non-critical logs only when verbose
d610a9c nic_router: multiple uplinks
75d9fe7 nic_router: get rid of the term "legacy"
486566c nic_router: optimize handling of invalid domains
65a5462 nic_router: fix DHCP-server deinitialization
9baa745 session_label: make typedef String public

m-stein, 2018-06-14:
@chelmuth I rebased and tried it out with the new Sculpt. This led to a small fixup (cc43c7f) that ensures the attachment of the initial uplink sessions to their domains and simplifies the Main constructor.

chelmuth, 2018-06-14:
@m-stein Thanks for the update! I'm busy merging your branch just now.

m-stein, 2018-06-15:
I fixed further details in the router on my way to fixing the "packet alloc failed" problem. As the last commit is a fixup for this issue I linked them all to this issue:
e71c166 fixup "nic_router: multiple uplinks" (handle_link_state)
3c86a03 nic_router: cancel remote ARP on new IP config
17e4247 nic_router: fix log when updating ARP waiters

commit 992e1ab1 (2018-06-14): nic_router: non-critical logs only when verbose

* Do not log events that are not critical (deadly) to the NIC router if not
  configured to be verbose,
* Print almost all log lines with a prefix of the domain name they are
  related to,
* And, do not use Genode::error and Genode::warning as they make it hard to
  read the log with the domain name prefixes.

Fixes #2840
commit 90fcba79 (2018-06-29): nic_router: non-critical logs only when verbose

* Do not log events that are not critical (deadly) to the NIC router if not
  configured to be verbose,
* Print almost all log lines with a prefix of the domain name they are
  related to,
* And, do not use Genode::error and Genode::warning as they make it hard to
  read the log with the domain name prefixes.

Fixes #2840
closed 2018-07-03 by nfeske