genodelabs/genode #2841: Hack'n'Hike 2019

m-stein, 2018-05-28:
The Hack'n'Hike 2018 just ended and we fully enjoyed the weekend again. There were a lot of projects going on like the exploration of ARMv8 boards, a lot of discovering and polishing the next Sculpt version with graphical UI, a graphical UI for Genode's counterpart to 'top', and many more ... . Furthermore we had relaxed barbeque evenings, an adventurous morning in a climbing forest, a hike through steep rocks and idyllic woods, and some lively games of table tennis.

That said, we definitely want to continue this tradition next year. In this issue we all can have discussions, ask questions, make plans, confirm participations, and get informed with regards to the the Hack'n'Hike 2019.

m-stein, 2018-05-28:
The idea came up to celebrate the next Hack'n'Hike at one of the many lakes in the Lausitz or in Brandenburg. I like this idea.

chelmuth, 2018-06-15:
I also like the idea to trade mountains for water activities. How about http://www.senftenberger-see.de/ ?

m-stein, 2019-01-07:
### Date and Accommodation

We finally found a date and an accomodation for the Hack'n'Hike 2019.
So, we'd like to invite you to let us know if you want to join us **till the end of February**.

#### Location

Ferdinands Homestay, Halbestadt 51, 01824 Königstein
- [Homepage 1 (en)](https://www.ferdinandshomestay.de/english/hostel/index.html)
- [Homepage 2 (de)](https://aktivreiseteam.de/hostel-im-elbsandsteingebirge.html)
- [Open Street Maps](https://www.openstreetmap.org/way/504526935#map=19/50.93732/14.07427)
- [Google Maps](https://www.google.com/maps/place/Ferdinands+Homestay/@50.9371009,14.0720842,17.75z/data=!4m12!1m6!3m5!1s0x4709a4b57a41d455:0xf116077a9d20132d!2sFerdinands+Homestay!8m2!3d50.9370414!4d14.0732812!3m4!1s0x4709a4b57a41d455:0xf116077a9d20132d!8m2!3d50.9370414!4d14.0732812)

#### Time
- Check-in as of 3 PM on May 10th 2019
- Check-out till 11 AM on May 13th 2019

#### Room
- We have booked the complete hostel with sleeping rooms for 16 participants for the time of the Hack'n'Hike
- One self-catering kitchen
- One seminar room

#### Misc
- Internet will be available
- Bedding will be provided

#### Catering
- The catering for breakfast by the hostel is included
- For lunch, we normally stop by a restaurant or tavern during the day trips
- The organization of the catering for dinner is not finished yet (the outcome will be part of the upcoming schedule)
- If you have special requirements regarding the food (vegetarian, vegan, gluten-free, lactose-free, ...), please **let us know as soon as possible**

skalk, 2019-01-10:
Please, count me in.

nfeske, 2019-01-10:
I'm on board. :-)

alex-ab, 2019-01-10:
I'm in.

jpouellet, 2019-02-06:
@nfeske thanks for the invitation - I would love to attend :)

nfeske, 2019-02-06:
@jpouellet That's fantastic! :-)

ssumpf, 2019-02-06:
Me and Pepino are in.

marmarek, 2019-02-15:
I'd love to be there too. Thanks for the invidation @nfeske!

rite, 2019-02-15:
@Sid (vegetarian) and I are in. Thanks for the invitation and organization!

nfeske, 2019-02-15:
@marmarek, @rite, @sid That's wonderful news! Welcome on board.

tomga, 2019-02-15:
I also would like to come, so count me in please.

ehmry, 2019-02-15:
@mato, can we confirm you?

mato, 2019-02-15:
@ehmry Yes, and thank you for the invite!

m-stein, 2019-02-16:
@mato, @tomga, @sid-com, @rite, @marmarek, @jpouellet 
Cool that you've joined the Hack'n'Hike :)

nfeske, 2019-04-01:
@m-stein I heard rumors that our accommodation has no internet connection. Maybe it would be a good idea to check back with the nice people from Ferdinands Homestay and, if there is indeed no internet, draft a plan to use LTE together with a wireless hot spot or something similar?

m-stein, 2019-04-01:
As far as I remember, they told me that there'll be an Internet connection (and I stated this also in "Date and Accommodation"). But I'll ask them again and then post an update.

m-stein, 2019-04-11:
I'd like to suggest a rough schedule and activities. Please feel free to suggest modifications!

#### Friday
**3 pm:** Arrival at the hostel "Ferdinands Homestay", Halbestadt 51, 01824 Königstein
**Evening:** Settle in and time for hacking
**6:30 pm:** 30 min. walk along the Elbe to the restaurant "Steingut"
**7:00 pm:** Dinner in the restaurant "Steingut"
**Night:** Walk back to the hostel and hacking
#### Saturday
**Breakfast:** Breakfast in the hostel (inclusive)
**10:30 am:** Hike to the rock formation "Gamrig" and the rock formation "Feldsteine/Honigstein"
**Lunch:** Lunch in a restaurant in Rathen (reservation pending)
**Evening:** Hike back to the hostel (optionally further hike to the rock formations "große/kleine Gans") and hacking
**Dinner:** BBQ by Genode Labs
**Night:** Hacking
#### Sunday
**Breakfast:** Breakfast in the hostel (inclusive)
**10:30 am:** Hike to the castle "Königstein" (1-2 € for the ferry)
**12:00 am:** Visit castle Königstein (10-12 € per person)
**Lunch:** Lunch in a restaurant in Königstein (reservation pending)
**Evening:** Hike back to the hostel and hacking (1-2 € for the ferry)
**Dinner:** Self-Catering by Genode Labs
**Night:** Hacking
#### Monday
**Breakfast:** Breakfast in the hostel (inclusive)
**11 am:** End of the Hack'n'Hike

chelmuth, 2019-04-11:
Thanks @m-stein for your thoughtful planning, I'm 100% with you. As it's only four weeks to the event, everybody should make travel plans as the venue is about 40 min walk (incl. ferry) from the train station in Königsstein.

m-stein, 2019-04-12:
**Unfortunately, there will be no Internet connection provided by the hostel!**
But they told me that, depending on your provider, there will be sufficient mobile coverage. Sorry for the inconvenience - the initial announcement was based on a misunderstanding. Maybe we should provide all contrib code locally and setting up a package server for the Hack'n'Hike.

gwitmond, 2019-04-13:
On the internet-situation:I  bring my phone with 5GB 3G/4G internet bundle to share. If there is sufficient reception...

On reception: I used to have this 3G wifi access point with external antenna, but that got stolen. The external antenna really made a difference between hardly any reception at all and blasting at full speed. If anyone has one, it might come handy :-)

On travel: I plan to take a two-day trip on the motorbike from NL with a stopover in near Kassel, anyone from that area coming as well?

m-stein, 2019-04-15:
### Journey Information
#### By bike ;-)
Like every year, I'll do both trips by bike and would be happy about companionship. I'm can provide to you a bike plus bicycle bags if you don't have your own. Please let me know if you like to join in ;)
25.05 outwards from Dresden Bahnhof Neustadt starting at around 10:30 am
28.05 returning from the hostel starting at around 10:30 am

#### By train
**25.05**

Station | Time | Platform |  Train
-- | -- | -- | --
Dresden main station | , Dresden main station - Großschönaudep 1:59 pm<br/><i>Every half an hour</i> | 18 | S1 direction Bad Schandau
Königstein<br/>(Sächsische Schweiz) | arr 2:37 pm | 2
Ferry near the train station<br/>to the other side of the river Elbe
To the left<br/>(down the river),<br/>always along the riverbank,<br/>35 minutes walk<br/>till the "Ferdinands Homestay"

**28.05**

Station | Time | Platform |  Train
-- | -- | -- | --
Up the river,<br/>always along the riverbank,<br/>35 minutes walk<br/>to the Ferry| best start at 10:15 am
Ferry Königstein<br/>to the other side of the river Elbe
Königstein<br/>(Sächsische Schweiz)  | dep 11:19 am<br/><i>Every half an hour</i> | 1 | S1 direction Meißen Triebischtal
Dresden main station | arr 11:58 am | 19

ehmry, 2019-04-15:
I would join by bike, and I am fully equipped.

m-stein, 2019-04-15:
@emery Cool :-)

gwitmond, 2019-04-15:
@m-stein 
> 25.05 outwards from Dresden Bahnhof Neustadt starting at around 10:30 am
> 28.05 returning from the hostel starting at around 10:30 am

That's from Saturday to Tuesday. Did you change the date from 10-13 may? Or did you copy the mail from last year?

Cheers ;-)

m-stein, 2019-04-15:
I've just fixed the wrong dates in the "Journey Information".

@gwitmond Thanks for the hint :-) It was indeed a copy-past error.

mato, 2019-04-16:
@m-stein Thanks for the detailed schedule. I plan to arrive via direct
train from Bratislava in Bad Schandau at 16:17, so will take a connecting
train to Konigstein if there is one, and/or walk directly to either the
hostel or Steingut, depending on train delays and weather on the day.

Regarding Internet, I have a fast LTE USB stick (Huawei E3372) I can bring
which works without drivers on anything that supports USB-attached NICs,
but would need someone to provide a suitable German SIM for it.

m-stein, 2019-04-16:
@mato 

Station | Time | Platform |  Train
-- | -- | -- | --
Bad Schandau | dep 16:11 | 1 | S1 direction Meißen Triebischtal
Königstein(Sächs Schw) | arr 16:15 | 1
-- | -- | -- | --
Bad Schandau | dep 16:45 | 3 | S1 direction Meißen Triebischtal
Königstein(Sächs Schw) | arr 16:48 | 1

I'll bring a (I think its micro) SIM that should have like 2 or 4 Gb per month that I don't use.

mato, 2019-04-16:
@m-stein

Thanks.

The Huawei takes a 2FF SIM (Mini-SIM), so bring the cutout.

cnuke, 2019-04-18:
@m-stein FYI, I will already leave on sunday which might influence the planning on your side.

marmarek, 2019-05-01:
@m-stein I'd join a bike too, I have everything needed.

m-stein, 2019-05-07:
According to the current weather report, we might have a wet weekend. So, it would be advisable for everyone to bring appropriate clothing. So, in case the rain doesn't get to heavy, we can do the hiking with an umbrella or a rain jacket.

m-stein, 2019-05-07:
@ehmry @marmarek: Regarding the journey by bike I'd like to wait until Thursday morning with a decision. Would this be ok for you  ?
If we would have to cancel the Friday bike tour because of the weather, we could easily fall back to the train and either take the bikes with us for a potential bike tour at Monday or leave the bikes in the Genode Labs office.

jpouellet, 2019-05-07:
I'll be arriving 2019-05-10 by multi-hop train with the following final segments:

| From (platform) | To (platform) | Time | on train |
|:-|:-|:-|:-|
| Hof Hbf (6) | Dresden Hbf (14) | 12:27 - 15:03 | RE 74039 |
| Dresden Hbf (18) | Königstein(Sächs Schw) (2) | 15:29 - 16:07  | S 1 |

in case anyone is coming from a similar direction / time and wishes to meet along the way.

sid-com, 2019-05-07:
@rite and myself will arrive in Dresden at around 0830 am. @m-stein is there anything we can do to help between our arrival in Dresden and the departure to Königstein?

m-stein, 2019-05-07:
@sid-com @rite: If you like, you can come to the Genode Labs office. Some of us will be there preparing the departure and having coffee rounds :-)

temp148937560, 2019-05-07:
In case you missed the news and bandwidth is still of some concern to you, Freenet yesterday started a flexible, unlimited data LTE plan, the first of its kind in Germany at that price point. SIM might still arrive before your departure, or you might be able to order it directly to your accomodation. Have fun!

https://www.heise.de/newsticker/meldung/Freenet-Funk-Mobilfunk-und-LTE-Flatrate-fuer-1-Euro-am-Tag-4414473.html

sid-com, 2019-05-08:
> @sid-com @rite: If you like, you can come to the Genode Labs office. Some of us will be there preparing the departure and having coffee rounds :-)

Sweet! Looking forward to it :-)

m-stein, 2019-05-09:
The weather doesn't look too bad tomorrow. So, I'll be in front of the "LIDL" supermarket of the train station "Dresden Neustadt" (east side of the building with the square and the parking area) around 10:30 am and hopefully find all the bikers there :-)

m-stein, 2019-05-09:
Of course, I'll be also in the Genode Labs office until 10:10 am.

m-stein, 2019-05-10:
@ehmry @skalk Marmek had to take a later train so he could be at Bahnhof Neustadt around 11 am. Is it ok for you if we start half an hour later? (If I don't get an ack from you I'll be at Bahnhof Neustadt around 10:30 am)

alex-ab, 2019-05-10:
@m-stein: @ehmry and @skalk are at the office and can start whenever you say

m-stein, 2019-05-10:
@alex-ab Thanks :-)

marmarek, 2019-05-13:
Thank you all for the fantastic weekend!

chelmuth, 2019-05-14:
A very insightful and enjoyable weekend indeed. Looking forward to #3349 in 2020.

closed 2019-05-14 by chelmuth

nfeske, 2019-05-14:
Thanks @m-stein for the splendid job with organizing the Hack'n'Hike and to all participants for making the weekend so enjoyable!

m-stein, 2019-05-14:
Thanks to everyone for this relaxing and inspiring weekend!

I'd like to encourage you all to present your personal HnH19 project, for instance, as article on [genodians.org](http://genodians.org/)! Please don't hesitate to share your story - even if it is a short one or still work in progress!

Hope, to see you again at the [HnH20](https://github.com/genodelabs/genode/issues/3349)!