genodelabs/genode #2846: chroot: 'merge' polices complex and redundant

ehmry, 2018-05-30:
The chroot server can perform partial conversion of sessions labels to root paths when a policy is configured with _merge_ and _label_prefix_ attributes. Implementing this requires some ugly string handling and is more or less redundant now that init can rewrite session request labels. The merge feature was added at a time when cli_monitor was used instead of dynamic init. This would chroot the children of cli_monitor into directories, any session of child X would be merged to /cli_monitor/X, child Y to /cli_monitor/Y, and so forth.

The following would chroot any session from the configured child to `/shared/x/y`, and that is powerful enough to satisfy the need for the merge behavior.
```
<service name="File_system">
  <child name="chroot" label="shared -> x -> y"/> </service>
```

ehmry, 2018-05-30:
Fixed at 6c30e986405f7c8db459f441615f10971d2e41f7.

I actually use a single chroot instance in my deploy and configure it using init, like this:
```
        <start name="chroot" pkg="ehmry/pkg/chroot/2018-05-30">
                <route>
                        <service name="File_system"> <child name="default_fs_rw"/> </service>
                </route>
                <config> <default-policy writeable="yes"/> </config>
        </start>

        <start name="vm" ram="5G" pkg="nfeske/pkg/vbox5-nova-sculpt/2018-05-22-b" version="1">
                <route>
                        <service name="ROM"    label="capslock"> <parent label="capslock"/> </service>
                        <service name="ROM"    label="platform_info"> <parent/> </service>
                        <service name="ROM"    label="usb_devices"> <child name="usb_devices_rom"/> </service>
                        <service name="Report" label="shape"> <parent label="wm -> wm -> vm -> shape"/> </service>
                        <service name="Nitpicker"> <child name="wm"/> </service>
                        <service name="Nic">       <child name="nic_router"/> </service>
                        <service name="Report"> <parent/> </service>
                        <service name="RM">     <parent/> </service>
                        <service name="Rtc">    <parent/> </service>
                        <service name="Usb">    <parent/> </service>
                        <service name="File_system" label="vm">
                               <child name="chroot" label="vm -> xubuntu"/> </service>
                        <service name="File_system" label="shared">
                                <child name="chroot" label="shared"/> </service>
                </route>
        </start>

```

nfeske, 2018-06-12:
I welcome this simplification as I never had a use for the "merge" feature.

BTW, you may remove the hash updates from the patch. The hashes are taken care of by our CI tools.

commit 69b25f91 (2018-06-14): Chroot: remove merge policies

The chroot server was conceived to automatically place File_system
sessions into segregated root directories by converting session labels
to paths. If multiple children needed to be grouped under the same path,
a 'merge' policy would truncate the session label before path conversion.
Now that init can rewrite session labels and thus reproduce truncation,
the chroot merge feature is redundant and can be removed.

Fix #2846
commit 9d8bf137 (2018-06-29): Chroot: remove merge policies

The chroot server was conceived to automatically place File_system
sessions into segregated root directories by converting session labels
to paths. If multiple children needed to be grouped under the same path,
a 'merge' policy would truncate the session label before path conversion.
Now that init can rewrite session labels and thus reproduce truncation,
the chroot merge feature is redundant and can be removed.

Fix #2846
closed 2018-07-03 by nfeske