genodelabs/genode #2848: foc + smp: component destruction renders kernel non-responsive

m-stein, 2018-05-30:
We discovered this while debugging the current problems with the ping tests. My simplified ping test has 2 ping components with a NIC session each to the nic_bridge (or the nic_router) running on foc x86_64 qemu with smp. As soon as one of the pings wants to do parent.exit() it seems that the kernel gets stuck at the end of Child::close_all_sessions(). The other ping proceeds a bit but at the latest gets stuck in parent.exit(). We observed that we can't get to the kernel debugger after this point anymore. It is the point where all the threads of the first ping got destructed and some of them may still be in a dead state. After some playing around, we found that disabeling smp in the kernel config fixes the problem. With smp enabled, we applied a patch of @cproc that flushes the TLB at some point in idle_op in foc but it didn't fix it. Now there is the question how to proceed with this:

* Dig deeper into the guts of foc to find the problem
* Disable or modify the tests that trigger the problem for foc to circumvent the bug
* Consider disabeling smp on foc until someone fixes the problem

alex-ab, 2018-05-31:
Just as note, it is a recurring issue in various shapes, see #2659 and #1495.

m-stein, 2018-05-31:
Thanks for the feedback!
@chelmuth So, I will do this.

m-stein, 2018-05-31:
Disabeling foc smp doesn't seem to be sufficient (btw. when changing the foc config it seems to be a good idea to disable or clear ccache). Although destructing a ping component now works well more often, it still crashes sometimes with different symptoms at the end of parent.exit(): 
```
KERNEL: Warning: CPU0: Pager of 1f is invalid (pfa=0000000000000000, errorcode=0000000000000010) to ffffffff81adc000 (pc=11451d3)
```

```
Bender: Hello World.
Need 00595000 bytes to relocate modules.
Relocating to 1fa4a000: 
Copying 3892016 bytes...
Copying 1046856 bytes...
Copying 738192 bytes...
Copying 167256 bytes...
```

```
General Protection (eip=f000d62cf000ef57, err=0000000000000000) -- jdb bug?
General Protection (eip=f000d62cf000ef57, err=0000000000000000) -- jdb bug?
General Protection (eip=f000d62cf000ef57, err=0000000000000000) -- jdb bug?
...
```

I admittedly play with the thought to simply disable foc for the ping tests.

chelmuth, 2018-05-31:
Go ahead, we have other platforms of greater relevance.

m-stein, 2018-05-31:
I continued a little bit looking into it. Apparently it can be prevented by skipping things in the destruction of platform threads or by simply skipping the destruction of the whole CPU session of ping. This way, I managed to circumvent the problem for x86 platforms but then it triggers other errors on ARM .

Thus, the simple approach:
ff33e92 ping.run: mark as "not supported" on Linux & FOC

closed 2018-06-13 by chelmuth

jermar, 2018-06-20:
The !in_sender_list() assertion failure might be fixed by c219876602f1cbb1fde93c85e771530364d72fb3

skalk, 2018-06-25:
@jermar thank you for the hint! I will test it soon.

jermar, 2018-06-25:
@skalk Let me know hot it goes. Looking into the latest changes, also other commits might be of interest.