genodelabs/genode #2850: Retire OMAP4 support

nfeske, 2018-06-02:
We introduced OMAP4 support in 2012. Back then, the Pandaboard was an attractive target for Genode because of the excellent documentation by TI, and the low cost and good availability of the boards. In the meantime, 6 years later, the OMAP line of SoCs has apparently faded into obscurity - at least from Genode's perspective. Given that OMAP support is not used by any customer of Genode Labs nor by any of the regular developers (well, except for the automated tests of course), keeping up our support for this SoC is questionable. Since the drivers are not stressed in practice, possible regressions may go unnoticed anyway. On the other hand, the maintenance of the drivers can become a burden at times. The latter is currently the case for @skalk's work on updating the DDE-Linux-based USB driver. He pointed out that further cultivating on the NXP i.MX series is a much better investment of our time than further nurturing OMAP4.

Should we better abandon OMAP4 to lift a bit of weight from our shoulders?

m-stein, 2018-06-03:
This issue is making me think of an easy way to browse abandonded features of Genode to prevent people from re-inventing the wheel and unnecessarily go through the pain that the initial developer had to face (which is especially true when it comes to hardware support). But its just a lazy-weekend-throught ;)

chelmuth, 2018-06-04:
AFAIK there's at least one active community member who uses a TI platforms. Bob Stewart repeatedly [posted](https://lists.genode.org/pipermail/users/2017-March/005084.html) on the mailing list about his port to TI AM437x/AM335x, but his developments are not accessible and a fork of base-hw anyway, so this may not be a reason to keep OMAP4 drivers.

N8Fear, 2018-06-04:
I'm porting Genode/seL4 to AM335x and as far as I can tell the MMC and GPIO drivers from OMAP4 can be used with somewhat minimal modifications.
I know I don't have a say anyways, but if one wants to maintain a port to another platfom they surely can support/maintain the drivers themselves and if you want to consider merging such a platform to the main repo the respective drivers could be (re-)imported also.
From my point of view Norman's original argument still holds.
What may be a good idea though, is documenting such removed "treasures" somewhere, so that anyone who wants to use (or bring back) the removed parts knows. that it may be a help to search the git history.

nfeske, 2018-06-12:
In an offline conversation, @skalk had the idea to move the OMAP4 drivers (like gpio, mmc, and uart) to genode-world. Still, we'd like to drop the support for USB because the driver would become unstable with our planned update of DDE-Linux und the it cannot easily be moved to genode-world anyway. @N8Fear would that be a good middle-ground for you?

N8Fear, 2018-06-12:
That would certainly work for me. But as stated above, I'm still in the process of finding out, if the drivers work unmodified or if I need to change them anyways.
If you want to clean the drivers from the tree just go ahead with it. I can keep them around myself and if I need them, I can use them on my branch. I don't thnk that keeping the drivers makes much sense if you're going to remove the platform. And should you need them in the future or if you consider to merge my work for AM335x, you can just just restore them.
In short: you don't need to keep them on my behalf.

nfeske, 2020-03-26:
Merged to staging

commit 433e1f0d (2020-03-26): Retire OMAP4 support

Fix #2850
commit f0510655 (2020-04-17): Retire OMAP4 support

Fix #2850
closed 2020-04-17 by chelmuth