genodelabs/genode #2851: Preserve hybrid MBR

cnuke, 2018-06-04:


alex-ab, 2018-06-04:
sculpt + bios/legacy now boots also after a expand on real hardware

chelmuth, 2018-06-04:
Beside this improvement, is it impossible for gpt_write to write a consistent hybrid mbr without any options/configuration requirements? Are there other use cases currently that are incompatible to this?

cnuke, 2018-06-04:
There is no technical reason, just that there is currently neither a heuristic for selecting a potential boot partition nor is there code for writing a hybrid MBR.

cnuke, 2018-06-13:
I am going to close this issue for now because the commits are in and how we want to deal with the hybrid MBR with regard to installing Sculpt on disk from within Sculpt should be the topic of another issue.

closed 2018-06-13 by cnuke