genodelabs/genode #2853: run: add '/sbin' to the 'PATH' environment variable

cproc, 2018-06-06:
On some Linux distributions, for example Debian, the '/sbin' directory is not included in the 'PATH' environment variable by default and tools like 'gdisk' are not found.

cproc, 2018-06-06:
Fixed

ssumpf, 2018-06-06:
@cproc: The `check_installed` function also searches /sbin.

`set gdisk_cmd [check_installed gdisk` in a run script should also work.]

nfeske, 2018-06-06:
@ssumpf good to know. That said, I find the following code more obvious

```
exec gdisk args...
````

compared to

```
exec [check_installed gdisk] args...
```

Don't you agree? Or do you see any potential problem with extending the PATH environment variable?

ssumpf, 2018-06-06:
@nfeske: Until now the workflow has:
`set gdisk [check_installed gdisk]`
`exec $gdisk args`
The other one is:
`requires_installation_of gdisk`
`exec gdisk`
I think we should get rid of one version. If the PATH way is preferred, we should also add '/usr/sbin' and '/usr/local/bin' since both have led to the same issue in the past.

nfeske, 2018-06-06:
Thanks for clarifying. I agree with everything you said. The former approach has the benefits of not relying on a global side effect (PATH env variable) and of doing the check implicitly (it does not require the discipline of adding the `requires_installation_of` call). Only the name of the function `check_installed` is unintuitive. It wrongly suggests that is returns the result of a check whereas it actually performs a lookup similar to the `which` command. Maybe the name `installed_command` would be a better fit? The following version looks more intuitive to me:

```
exec [installed_command gdisk] args...
```

chelmuth, 2018-06-06:
It's worth to mention that `check_installed` always exits with failure if the tool is missing. *tool/run* also provides `have_installed`, which returns true or false and currently does not check the additional paths.

chelmuth, 2018-06-14:
@ssumpf any updates regarding this issue?

ssumpf, 2018-06-14:
@chelmuth: Not yet, but I have it on my TODO list.

ssumpf, 2018-06-19:
@chelmuth: Please have a look at 4a7712f if this is the right path to proceed.

chelmuth, 2018-06-19:
After a quick look I like it - will review if you're finished.

ssumpf, 2018-06-20:
@chelmuth:  d62f388 and  3667454 should resolve the issue.

chelmuth, 2018-06-21:
@ssumpf  Could you please rebase d62f388bbec430b3039ddb52cfe260643f6d8477 onto staging (resp. 2e1e9241611f9ad14d43e577ca7725ea55c0173f) as there some non-trivial merge conflicts.

ssumpf, 2018-06-22:
@chelmuth: Done with a889e3ec4e7110e0f17bc3b3711b8bb781c1c531

commit 8fc651fb (2018-06-22): run: rename 'check_installed' to 'installed_command'

Also remove 'requires_installation_of', while also checking sbin
directories in 'have_installed'. The run scripts have been adjusted
accordingly.

Fixes #2853
commit 10485b0d (2018-07-03): run: rename 'check_installed' to 'installed_command'

Also remove 'requires_installation_of', while also checking sbin
directories in 'have_installed'. The run scripts have been adjusted
accordingly.

Fixes #2853
closed 2018-07-03 by nfeske