genodelabs/genode #2855: depot_query: handle invalid runtime config

rite, 2018-06-07:
If a (to be deployed) package provides an invalid runtime config, depot_query may end up in an endless loop or page fault.

alex-ab, 2018-06-12:
5b1112ef558d86b5bf386243a794b876fadf723e fixes the page fault for me.

@nfeske: Nevertheless depot_deploy generate a init config without the the missing start node. The code looks like that this is intended, neverthless the other start node in init fails since the missing start should provide a service ... In 2776961ecbf71efe5f75a1ceb2ab3161c58b4ba8 I changed to generate the init config solely if all nodes are complete, but as said, I don't know whether this is intended.

nfeske, 2018-06-12:
@alex-ab thanks a lot for the added exception handling! I don't have a good answer to the question how depot_deploy / sculpt_manager should deal with such situations. In the sculpt case, we should be fine because the sculpt manager performs an inter-child-dependency check before generating a start node. We will likely get a message at the "Diagnostics" window that the dependency is missing. This should raise eyebrows when a start node is actually there.

That all said, I'm very happy that the depot-query tool has now become robust against malformed input. Thank you!

commit 38ec9f0a (2018-06-12): base: reset unfinished xml nodes in xml_generator

Fixes #2855
commit 7d435c28 (2018-06-13): base: reset unfinished xml nodes in xml_generator

Fixes #2855
commit ef482513 (2018-06-13): base: reset unfinished xml nodes in xml_generator

Fixes #2855
closed 2018-06-13 by chelmuth