genodelabs/genode #2857: nic_router: fix uncaught interface RAM exhaustion

m-stein, 2018-06-08:


m-stein, 2018-06-08:
Fixed:
4d379ae autopilot.list: add nic_router_flood
6c594e6 nic_router_flood.run: test DoS against router
79dc198 nic_router: fix uncaught interface RAM exhaustion

m-stein, 2018-06-11:
@chelmuth I added a fixup that should fix the page faults with NIC router after applying the fix for this issue:
c63a4dc fixup "nic_router: fix uncaught interface RAM exhaustion"

m-stein, 2018-06-11:
I played around a while with sculpt without problems before uploading the fixup.

chelmuth, 2018-06-15:
The flood test is broken.
```
checking configuration syntax
  CHECK init
  CHECK init -> flood_links_tcp
  CHECK init -> flood_links_udp
  CHECK init -> flood_links_icmp
  CHECK init -> nic_router
  CHECK init -> ping
.config_0.xml.tmp:1: element config: Schemas validity error : Element 'config', attribute 'period_ms': The attribute 'period_ms' is not allowed.
.config_0.xml.tmp fails to validate
Error: Invalid XML syntax
```

m-stein, 2018-06-15:
@chelmuth I've added two fixups:
38b9603 fixup "nic_router: multiple uplinks" (nic_router_flood.run)
e346cd1 fixup "nic_router_flood.run: test DoS against router" (period_ms)

m-stein, 2018-06-19:
@chelmuth This should fix run/nic_router_flood on kernels other than nova:
6d9c5bf nic_router_flood.run: be more flexible
daf03c3 nic_router: verbose_packet_drop attribute

nfeske, 2018-07-03:
The commits entered the master branch today. @m-stein Can we close this issue or are you planning further supplemental commits?

m-stein, 2018-07-03:
No, I think this is it.

closed 2018-07-03 by m-stein