genodelabs/genode #2862: run: use sgdisk instead of parted in image/disk

rite, 2018-06-11:
As reported on the mailinglist, there seem to be some issues regarding the use of parted in run the scripts. All of them are solved and the patches are already applied to staging.

That said, I encountered another issue that only occurs when building the sculpt image within an Ubuntu 16.04 docker container - and only if it's run on GitLab CI... After some inconclusive investigation I decided to give sgdisk a try. It works on my archlinux and on the build server. Last but not least, scripted partitioning using sgdisk feels less awkward than using parted.

What do you think of replacing parted by sgdisk?

chelmuth, 2018-06-12:
Thanks for the patch. It looks pretty clear now which steps are conducted and only one tool is used for image generation. It seems worthwhile to extend the adaption to *image/uefi* though.

rite, 2018-06-12:
@chelmuth, please have a look at 81090d5. According to `sgdisk --info` and `parted print`, the partition table created with sgdisk look the same as those created with parted.

nfeske, 2018-07-03:
Fixed in master.

closed 2018-07-03 by nfeske