genodelabs/genode #2866: base: remove first-class support for statically-linked binaries

nfeske, 2018-06-12:
During the past few days of stress-testing Sculpt TC, the interaction between the runtime-init and the depot-rom (fs_rom) component (running as a child of the runtime-init) was troublesome in several situations. Normally, we have the golden rule that parents should not depend on their children. But here we have an exception that I was willing to accept because it simplified so many other aspects of the system. I figured that, ultimately, init needs to get hold of the ELF binaries to load. Since the ELF binaries are accessible by services on top only (the file system and the depot-rom), the runtime-init has to depend on this chain of ELF-binary-providing components. So the depot-rom should be thought of being a plugin of init rather than an untrusted child component.

However, the downsides are not easy to swallow:

* There is a possible livelock when the runtime-init calls `Rom_session::dataspace` and the fs_rom establishes a new session as a side effect during this call (workaround at https://github.com/genodelabs/genode/commit/37be984d347bf4dad0f58f470312bd3f3b850c9a)
* There is another possible livelock when `fs_rom` issues a resource request to init while dispatching the `dataspace` RPC (workaround at https://github.com/genodelabs/genode/commit/47ab45fddc435c70fb477cafb512c7fcf5c5a370)
* More possible livelocks are on the horizon once we switch fs_rom to a truely asynchronous session creation, as @ehmry is exploring in https://github.com/genodelabs/genode/issues/2760
* There is a hard dependency of the runtime-init's health from the depot-rom and the used file system, which also effects components that are already running and are not in any way depending on any file system.

After discussing various options for decoupling the relationship between the runtime init and the depot-rom with @chelmuth , it occurred to me that the only reason we have this hard RPC-dependency is that init looks into the ELF binary to figure out whether it is dynamically or statically linked. In the former case, it loads the dynamic linker. Today, this is the most prominent - if not to say the almost only - case. If we remove this check and instead always request `ld.lib.so`, the runtime init could fetch this binary always from the trusted parent (the static boot image). All other ROM sessions are then issued by the dynamic linker from within the running component.

In the rare cases where we want to load a statically linked component, label-rewriting route like this would do the trick:
```
<start ...>
  <route>
    <service name="ROM" unscoped_label="ld.lib.so">
      <parent label="statically_linked_executable"/>
    </service>
  </route>
</start>
```

This would nicely sidestep a lot of trouble and - at the same time - simplify the child construction. What do you think?

@alex-ab I am aware that the NOVA platform test and Seoul are statically linked. Would that be show stoppers for the above plan? Or does anyone else know of other components that rely on the current first-class support for static binaries?

nfeske, 2018-06-12:
Yup. That's the idea.

alex-ab, 2018-06-12:
Ok, seoul is special any way and having a special route is not making it less special anyway ;-)

commit 08b94a7e (2018-07-04): base: rm first-class support for static binaries

This patch removes the detection of statically linked executables from
the base framework. It thereby fixes the corner cases encountered with
Sculpt when obtaining the binaries of the runtime from the depot_rom
service that is hosted within the runtime.

Statically linked binaries and hybrid Linux/Genode (lx_hybrid) binaries
can still be started by relabeling the ROM-session route of "ld.lib.so"
to the binary name, pretending that the binary is the dynamic linker.
This can be achieved via init's label rewriting mechanism:

  <route>
    <servic [... cut]
commit 4b46abf8 (2018-08-02): base: rm first-class support for static binaries

This patch removes the detection of statically linked executables from
the base framework. It thereby fixes the corner cases encountered with
Sculpt when obtaining the binaries of the runtime from the depot_rom
service that is hosted within the runtime.

Statically linked binaries and hybrid Linux/Genode (lx_hybrid) binaries
can still be started by relabeling the ROM-session route of "ld.lib.so"
to the binary name, pretending that the binary is the dynamic linker.
This can be achieved via init's label rewriting mechanism:

  <route>
    <servic [... cut]
closed 2018-08-08 by nfeske