genodelabs/genode #2868: nit_fb: allow sizes to be specified in percent of the screen size

trimpim, 2018-06-15:


nfeske, 2018-06-15:
This change moves nit_fb to a direction that should better be addressed by a proper window layouter. Even though it makes nit_fb more flexible, it still won't be flexible enough. I predict that one would next desire features like:

* Positioning the nit_fb view as a screen-resolution-relative percent value.
* Specifying a pixel gap to be applied in addition to percentage values.
* Support percentage values smaller than 1%.

However far we go, there will always be one feature missing. Each of those additions increases the complexity and test surface of the component while the overall direction remains a dead end.

Wouldn't it be better to use the window manager and a custom window layouter instead? At least this was my line of thinking then designing the leitzentrale GUI of Sculpt TC. Here, the sculpt manager plays the role of a window layouter and is thereby able to apply a layout policy that would be impossible to achieve by merely using nit_fb  - like stacking the log and inspect windows vertically or horizontally depending on the available screen real-estate, or aligning the left border of those windows with the right border of the menu regardless of the menu's font size.

For reference, you may take a look at the commit https://github.com/genodelabs/genode/commit/3372c1a7b24ba617c90307c32711d11d4d44d905 that implements this technique to sculpt's leitzentrale. The interesting part is the `_handle_window_layout` method.

trimpim, 2018-06-15:
I see your reasoning, and agree with you that a proper window layouter is the way to go.

So I close the issue and create a other for the addition of the parser for quantities.

closed 2018-06-15 by trimpim

trimpim, 2018-06-15:
I think I can live with keeping quantity in my own component. The reason why It ended up in `util/string.h`is, that I followed the `Xml_attribute.value()` to where it parsed `Number_of_bytes`.
If I understand it correct, I could do it as it is done in `./repos/os/include/nic/xml_node.h` where the `ascii_to()` function to parse MAC addresses is. The class Quantity can then even be placed in to an other namespace.

nfeske, 2018-06-15:
Yes, that's good.