genodelabs/genode #2875: libc: socket peeking

jklmnn, 2018-06-20:
While using the libc `recv` call with the `MSG_PEEK` flag I noticed that it isn't supported and according to the mailing list this also won't be supported in the future.

An idea would be a special file in socket_fs as suggested by @ehmry.

chelmuth, 2018-06-20:
I suggest we broaden the discussion more in the direction: Should we really provide `MSG_PEEK` in the socket-fs or could those use cases be fulfilled by other means? I'm a bit hesitant to jump right into designing new POSIX-influenced features as POSIX definitely has its dark corners (think of `ungetc()`).

(Original discussion started at the [mailing list](https://lists.genode.org/pipermail/users/2018-June/006128.html).)

jklmnn, 2018-06-20:
Of course we can broaden the issue. `MSG_PEEK` was just the example to start the discussion (I could have phrased it better). My initial problem was a mechanism to react on available bytes on a socket and transform this into an asynchronous API.

ehmry, 2018-06-21:
I have a simple PEEK implementation on my `issue2875_peek` branch, I'm a bit on urlaub, so not yet well tested.

chelmuth, 2018-08-02:
PEEK already sneaked into staging with #2335, hm.

ehmry, 2018-08-02:
Yes, if you know the file is there and how to use it.

cproc, 2020-04-28:
qtwebengine uses this feature as well (to check if a socket is connected). I rebased @ehmry's commits to staging and made some corrections in the lxip implementation and checked with a modified version of the 'libc http client test' [1] and it appears to work now with both lwip and lxip.

[1] https://github.com/cproc/genode/commit/d73bf77

chelmuth, 2020-04-29:
What a pity but at least it's just the legacy world. Any chance to support MSG_PEEK in lwip too?

cproc, 2020-04-29:
@chelmuth: lwip has the peek file already

commit e8a0dd26 (2020-05-11): Libc: MSG_PEEK support

As discovered by Johannes Kliemann, peeking at buffered socket data
using 'recv' and 'MSG_PEEK' is not supported. Read a "peek" control file
from the socket directory to attempt to peek into buffers at the
socket_fs. Support for every feature of POSIX sockets cannot be
expected, but this one is trivial to implement.

Fix #2875
closed 2020-05-18 by nfeske