genodelabs/genode #2876: log interface floating point output error

al-sharp, 2018-06-20:
Sometimes a double value is rendered strangely (by source in out_float) in log output. (Genode 18.05, x86_32)  E.g.
```
    unsigned N1 = 3300;
    unsigned N2 = 5000 - N1;
    double div1 = N1 / 1000.0;
    double div2 = N2 / 1000.0;
    log(N1,"/1000.=",div1,", ",N2,"/1000. =",div2)
```
```
    [init -> test] 3300/1000.=3.299999, 1700/1000. =1.6a0
```
Note especially the "1.6a0".

Possibly some of the code here: http://www.netlib.org/fp/
e.g. g_fmt, could be adapted--it is not very long but license compatibility uncertain.
Reference:   https://ampl.com/REFS/rounding.pdf

chelmuth, 2018-06-21:
I traced this to https://github.com/genodelabs/genode/blob/master/repos/base/src/include/base/internal/output.h#L136 where the first decimal place of `1.7` gets rounded to `6` and the second to `10` (resp. `0xa`). I ported the code into a Linux program but this did not show this error with GCC 5.4.0-6ubuntu1~16.04.9 and *gcc -m32*. Unfortunately I lost the Linux code (in */tmp*) after reboot.

cproc, 2018-12-10:
In this specific test case 6.999... is casted to 6 and then this 6 is supposed to be subtracted from 6.999..., but the compiler decided to move the 6.999... from the FPU to the stack and back in the meantime, which changes it into a 7 because of different encoding formats in the FPU and in memory, so the 6 is subtracted from 7 and the next decimal place becomes a 10.

By copying the original floating point value into a volatile variable and calculating with that variable, the conversion step between FPU and stack becomes more deterministic and improves the consistency of the output.

commit ff709cfe (2018-12-11): base: improve floating point output

Fixes #2876
closed 2019-01-07 by nfeske