genodelabs/genode #2878: net: handle Ethernet FCS

m-stein, 2018-06-21:


m-stein, 2018-06-21:
I've written an implementation for FCS updating/checking by looking at what the Wireshark source code does. But, I must admit that I have doubts whether my initial idea - that software considers the FCS - makes sense. I gathered some information that may underline my suspicion.

My setup is as follows:
```
wireshark --- L) lx ip/nic --- phys net --- N) ported ipxe nic --- R) nic_router --- P) native ping
                                            W) ported lx wifi      B) nic_bridge     T) test using ported lwip
                                                                                     V) virtualbox driving lx
```

So far, B/R do not check/update the FCS.

* If P sends a packet that ends at the end of the IP portion (no space for FCS), wiresharks capture doesn't contain an FCS.
* If P sends a packet with space for the FCS but FCS = 0 (or any other incorrect value), wireshark gets it and complains.
* The replies that P gets from L show no space for FCS, neither in wireshark nor in R/B.
* But, if P sends a packet with a correct FCS and R/B updates it, wireshark receives it and confirms that it is correct.
* If V or T send an IP packet, both R and wireshark show that it ends at the end of the IP portion (no space for FCS).

It looks like neither L, N, W, T nor V consider an FCS at all. It might be the case that the physical interfaces add an FCS at the end of the frame that is truncated again later by the remote hardware.
In this case, a software-FCS would be tunneled by the hardware as frame padding and Wireshark thinks it is the "real" FCS.

That said, my proposed way how to treat the FCS in Genode would be:
Leave it up to the physical interfaces. If possible and desired, let drivers configure hardware to check/update it. Other components would continue ignoring it.

The problem that software like Wireshark complains about our packets seems to originate only from the fact that our packets are bigger than needed. This is because, in some places, we allocate an oversized packet as the real packet size is hard to foresee. We could try to fix this.

The solution that would implement my initial idea:
Packet forwarders detect for each packet wether there is space for the FCS, and if so, check/update it. Packet generators could be configurable whether to add the FCS or not.

But, there is a problem with this. It seems that components cannot detect whether an FCS exists without parsing the packet (for some Ethernet sub-protocols only the Ethernet header for others also the sub-protocol header). This seems to be the same in Wireshark, there are comments in the code like "If we don't know whether the FCS is present, ..." . That said, there seems to be no bullet-proof implementation for this solution.

Any opinions?

chelmuth, 2019-03-14:
I suggest to not pursue this road and close the issue.

m-stein, 2021-01-19:
I'd finally close this issue. Sorry for delayed reaction.

closed 2021-01-19 by m-stein