genodelabs/genode #2880: grub2 update breaks vga character framebuffer

jklmnn, 2018-06-22:
https://github.com/genodelabs/genode/commit/193e14a489ff663ce3039244c8864c811a558e58 breaks the type 2 VGA character buffer that was usually provided by NOVA on legacy boot.
This information was usually provided instead of the framebuffer information for the `fb_boot_drv` in the `platform_info` rom but is missing since this commit.
Would there be any option to reenable this (at least with a flag)? I often need this framebuffer to provide a log session on platforms that have no other easy way of logging.

alex-ab, 2018-06-23:
I did not expected that anybody is interested in vga text mode for Genode/NOVA. With 441d67a22af43204bed36838023f703c2bce2fb3 the text mode information gets propagated again. 1ae2e1c7c1ac13b574762da7ce0498d82aff6048 showcases how to switch. Does it suffice your use-case ?

jklmnn, 2018-06-23:
These two commits fix the issue. From your comment I assume that only https://github.com/genodelabs/genode/commit/441d67a22af43204bed36838023f703c2bce2fb3 will go upstream (which would be fine for me)?
I use a [VGA LOG session](https://github.com/jklmnn/genode-trabant/tree/vga_scrolling/src/drivers/vga_log) for debugging that supports colors and scrolling (if you attach it to an Input session) up to 1024 lines.

alex-ab, 2018-06-25:
Nice tool, good to know ! Yes, I would like just to get 441d67a22af43204bed36838023f703c2bce2fb3 merged.

alex-ab, 2018-06-25:
And it is using Ada, very cool !

nfeske, 2018-07-03:
The fix entered the master branch.

closed 2018-07-03 by nfeske