genodelabs/genode #2884: foc: update to most recent kernel version

skalk, 2018-06-26:
Quite recently, there was some movement inside the Fiasco.OC development to target the kernel object destruction problems we encountered for a long time. Thanks to Jakub @jermar for hinting us!
I've tested it, and it seems promising. Therefore, we shall raise the kernel version used by Genode.
Ref #1495
Ref #2659
Ref #2848

jermar, 2018-06-26:
Hm, r79 does not contain all the potentially interesting fixes as it is older than the version on GitHub.

skalk, 2018-06-26:
Well, in fact the commit updates to r79 + all recent Fiasco.OC commits from the Github repository.

skalk, 2018-06-26:
Genode now references this Fiasco.OC/L4Re version: https://github.com/skalk/foc/commits/r79

commit b07591fc (2018-06-26): foc: update to recent revision r79 (fix #2884)
commit b844c659 (2018-06-28): fixup: "foc: update to recent revision r79 (fix #2884)"
commit 41fdb1d3 (2018-07-03): foc: update to recent revision r79 (fix #2884)
closed 2018-07-03 by nfeske