genodelabs/genode #2889: Expose raw input event data for non-C++ clients

ehmry, 2018-06-27:
Handling input events with Nim is cumbersome because lambda code must be injected to access event attributes or an alternative input event header must be maintained (what I am doing now). I would prefer to have a base class of event that I can use to access the raw implementation of the input event buffer. 

The same safety measures for interpreting events will be enforced in Nim, but that implementation  requires raw access.

ehmry, 2018-06-27:
Patch at dc34eb7668387450f300fee6bcd8b768791a3384.

nfeske, 2018-06-27:
Is there really no way around that? Your suggestion defeats the purpose of my information-hiding attempt. My undertaking was motivated by two goals, safety and the flexibility to change / expand the internal representation of the events in the future. I left the enum deliberately out of the public view. Exposing the internal representation opens new ways to work around the safe interface, and limits our ability for future changes. It would be a crack in the API I was happy to have closed.

I wonder, what speaks against a function that takes an Input::Event and extracts your custom representation in a format of your choice by calling the existing interface methods? In my opinion, one such (maybe awkward?) function hidden in a non-central place (the Nim backend) is better than weakening the public Genode API.

ehmry, 2018-06-27:
The enum can still be hidden, and I would not consider `Event_base` as part of the Genode API, just a part of the Input session definition. The problem is that a typed union is idiomatic for Nim, so I can only imagine to call every accessor to recreate the union, which is more complicated than just maintaining a header that defines a binary compatible structure.

ehmry, 2018-06-28:
@nfeske I have a patch to add a `Bindings` hatch as you recommended at d5297f4c28c90b95aff957d5fbc714033b7f8260. This is a much more innocuous solution, and makes unsafe access explicit and easy to discover.

nfeske, 2018-06-28:
I like this much better. @ehmry thanks for reworking the patch.

@chelmuth What do you think? Is the name `Bindings` for the hook fitting?

ehmry, 2018-06-28:
I like `Input::Binding`, fixup at cf8ab5a162db72a92a8258d5fedade3cdfe29a33.

nfeske, 2018-06-29:
Merged to staging.

ehmry, 2018-06-29:
My input bindings are published here: https://github.com/ehmry/nim-genode/blob/master/src/genode/inputs.nim, only symbols marked with `*` are exported.

commit 86736def (2018-06-28): Input::Binding for non-C++ language bindings

Quietly insert forward declaration of a Input::Binding class, and make
it a friend of Input::Event and Input::Session_client. This is to allow
non-C++ language bindings (Nim) to access private members by providing
their own implementation of the Binding class.

Fix #2889
closed 2018-07-03 by nfeske