genodelabs/genode #2890: sculpt: manual assignment of USB devices to clients / VMs

nfeske, 2018-06-27:
In the current version of Sculpt, the driver manager automatically configures the USB driver such that USB storage devices can be assigned to VMs. However, this built-in policy does not cover non-storage devices. To make Sculpt more flexible, it would be nice to give the user a way to manually express a device-assignment policy.

nfeske, 2018-06-27:
Commit be52d806595f0069875a5945d95dfa35dcd9e43b adds the policy hook.

nfeske, 2018-06-27:
Actually, I just noticed that the example entry of the usb policy cannot not work because - unfortunately - the teensy microcontroller used in the ergodox reports being a HID device. As Sculpt uses the USB driver's built-in HID support, the device is always claimed by the USB driver internally. So we might better use another example (like an USB-serial adapter).

ehmry, 2018-07-02:
Yes, this is also a problem with my Bitcoin hardware wallet, HID seems to be the recommended abstraction for USB devices now.

nfeske, 2018-08-08:
Fixed in master, albeit not generally useful until we remove the current eager grabbing of HID devices by the USB driver. Unfortunately, a lot of devices (like arduino device) present themselves as HID device, which makes it practically impossible to assign it to a VM. However, reworking Sculpt's USB driver approach is a different issue.

closed 2018-08-08 by nfeske