genodelabs/genode #2891: Signature checking fails if signature was created with a subkey

ValiValpas, 2018-06-27:
The component `app/verify` deals with signature checking based on GnuPG. Unfortunately, the current implementation only reads the first packet from the `pubkey` file, which contains the primary key. In consequence, depot files cannot be signed with subkeys.

The idea behind subkeys is that you can use the same public key across several devices but only store a dedicated secret subkey on each device. If one device gets lost or compromised, this only compromises the particular subkey.

nfeske, 2018-06-27:
The verify tool is a wrapper around the GnuPG code. Are you planning to have a go at this feature? I ask because I'm somewhat hesitant to dive into GnuPG internals anytime soon.

ValiValpas, 2018-06-28:
I first went for the quick solution and created a new keypair :(

Nevertheless, your comment motivated me to have a deeper look and I came up with a solution which I just pushed.

nfeske, 2018-06-29:
@ValiValpas I'm in awe! :-)

Have you considered adding the sub-key handling to the test? I don't insist, but I think it should be not too hard to add.

ValiValpas, 2018-06-29:
Done!

nfeske, 2018-07-03:
Thanks a lot. The test pattern is quite fitting for this issue. I merged both commits to staging now.

ValiValpas, 2018-07-04:
@nfeske  It seems that the genodelabs/pubkey has an invalid self-signature on the user id. At least this is what I got from the verbose info messages.

nfeske, 2018-07-05:
I have to admit that this goes over my head.

Having slept again over it, I think we may better not force it. I already felt uncomfortable with the staggering complexity of GnuPG. Your feature addition doesn't make it any less complex while possibly complicating a future transition to another implementation on OpenPGP (neopg, or sequoia-pgp?) since we would need to preserve the feature set.

GnuPG is a strange animal. It's probably better to not poke it with a stick too much. I guess that you - after having implemented your patch - can share my sentiment. I'm feeling bad because you put so much effort into it but would it be acceptable for you if I revert the commits from staging?

ValiValpas, 2018-07-05:
For the record: It seems that there is a problem with checking the signatures for the user id with the error code GPG_ERR_DIGEST_ALGO. This is because the genodelabs/pubkey uses SHA512 which is currently not compiled into libgcrypt. If I add this, the key works.

@nfeske , yes I agrre with your statement that GnuPG is a strange animal. I have seen to much of the codebase to feel confident with using GnuPG at all. It's just full of side effects, multiple implementations of similar functions in different flavours, etc.

Since this patch enables all kinds of validity checks of GnuPG (signatures, keys, revocation, expiration, etc.), it may render the verify component less robust as it increases the number of corner cases. I like the simplicity of the former approach. If I had known the complexity and implications of this patch before, I wouldn't have spent any effort.

nfeske, 2018-07-05:
@ValiValpas thanks for your understanding. I'm happy that we are on the same page.

cnuke, 2018-07-05:
FWIW, I also encountered the problems with regard to using subkeys but took the easy way out and opted for creating a new private key just for the use within the scope of the depot. One does not gain much by having the depot user's UID attached to its normal OpenPGP identity IMHO. For the time being it might make sense to treat the use of GPG as signature check-tool purely as an implementation detail.

commit 77ecc993 (2018-07-03): Signature checking: add subkey support

Fixes #2891

I moved the emulation one layer deeper into gnupg such that we
now emulate the `keydb_` functions.
This way, we can use the original implementations in
`import.c` and `getkey.c` for reading keyfiles and finding keys.

Note that I modified the gcc options to avoid implementing
many dummy functions from `keydb.c` which are never called.
commit a67d5169 (2018-07-03): fixup! Signature checking: add subkey support

Fixes #2891
closed 2018-08-08 by nfeske