genodelabs/genode #2897: init: unexpected element 'report' when validating against config.xsd

rite, 2018-06-30:


nfeske, 2018-07-02:
Good catch! Thanks for the patch. It misses a few attributes though. The full list of supported options is as follows (each is a boolean):

```
ids
requested
provided
session_args
child_ram
child_caps
init_ram
init_caps
```

rite, 2018-07-02:
@nfeske, yeah, I didn't check carefully enough ;) Commit 99cedeb adds all attributes.

Regarding your hint about the attribute types: 70049de63f04ab6f0a197f4eedb70752094503ee adds a boolean type definition that reflects the genode XML attribute parsing for boolean. In order to add some real benefit - in contrast to this commit - the Genode specific types coud be defined in an non-component specific schema within the Genode namespace. Further types could be RAM quantum or archive name. But I'm unsure if it's worth the effort...?

nfeske, 2018-07-03:
Thanks for updating the patch. I merged 7e49e00898bc7ca421fe4404172abd10ffaec438 to staging.

@m-stein What do you think about @rite's suggestion to distinguish different argument types that follow the same conventions throughout Genode?

m-stein, 2018-07-04:
My suggestion:
68814d9 base: define generic config XSD types globally

* I'm not sure if the path of the "genode" config inc is what one would expect it to be.
* The "genode" config inc contains all types that are currently used in multiple config XSDs. I don't know whether to split it up because some types are not relevant to base like 'Ipv4_address' or 'Port'. If so, my suggestion would be to name the XSD "env variables" like BASE_CONFIG_INC, NET_CONFIG_INC, ... and so on.
* I have replaced all types that I were able to replace without extra effort in the init config. There are further attributes that might get typed more restrictively.
* By now, the Number_of_bytes type is pretty loosely defined.

rite, 2018-07-04:
@m-stein, that's great! As usual you guys are at least one step ahead ;)

In adf46ce, I tried to get around the mechanism of writing a placeholder into the XSD which then needs to be replaced by the run tool. Instead a lookup path to "load DTDs or entities" is provided when calling xmllint. This way, the XSDs are valid "as is" and don't need to be patched before using them for XML validation.

m-stein, 2018-07-04:
@rite Great fixup! I didn't know that --path can be used for this :)

I added further fixups:
 88c15f7 fixup "base: define generic config XSD types globally" (include per lib)
0e05cd5 fixup "base: define generic config XSD types globally" (move Trace_policy_name)
4dd57af fixup "base: define generic config XSD types globally" (rename config.inc base_types.inc)
76a0be1 fixup "base: define generic config XSD types globally" (remove genode_dir_esc)

@nfeske @rite What do you think about it?

rite, 2018-07-04:
I just thought about checking XML files from recipes. I.e. #2894 adds an automatic check of runtime files. I'm unsure if the definition of the XSD types have some impact?

rite, 2018-07-05:
Never mind... XSD are not considered in #2894. XML validation using XSDs in recpies can be added later, if needed. I'm happy with your solution @m-stein

nfeske, 2018-08-31:
Now that d2a36c595842e4ede4b232c1c88480ef8b75f3a9 entered the master branch, we can close this issue.

closed 2018-08-31 by nfeske