genodelabs/genode #2898: xml_node: add content getter methods

insidioussid, 2018-07-02:
For 3rd party tools it is quite common to have information stored as xml content rather than xml attributes:

`<dofancystuff>false</dofancystuff>`

nfeske, 2018-07-02:
I wonder, couldn't that feature be implemented using the existing accessors, outside the `Xml_node` class? Throughout Genode, we usually keep such supplements local to the components that need them. E.g., for the Sculpt manager, I created several utilities in https://github.com/genodelabs/genode/blob/master/repos/gems/src/app/sculpt_manager/xml.h but I keep them local the the component. In particular the `query_attribute` may be generally useful but its not important enough to be added to the base API.

I explained my hesitant position about extending the API in https://github.com/genodelabs/genode/issues/2868#issuecomment-397605000

Following the line of thinking expressed there, the base API should be extended only if the base framework benefits from the change - if the overall complexity goes down. In contrast, your suggestion would extend the API solely for the convenience of applications / 3rd-party tools developed on top. Hence, it is better to keep the feature outside the scope of the base API.

BTW, speaking of interfacing with 3rd-party tools, have you considered using a full-fledged XML library instead of Genode's limited utilities? E.g., the Genode-world repository (https://github.com/genodelabs/genode-world) features a port of libxml2. Admittedly, I have no experience using this library but it seems to be the go-to solution for parsing XML.

insidioussid, 2018-07-04:
I see that there is a critical trade off between features and keeping the complexity of the base API as low as possible. I completely understand your line of thinking about extending the API in #2868. However, given the fact that information in xml can be placed in attributes or in content and that the latter is quite common, I would have expected that the two proposed functions would be closer to the border of the trade off mentioned above than #2868. 
For our use-case using a full-fledged xml library like libxml2 just to be able to get content would be too much. 
So, I will go with the option to create an utility like you did for `query_attribute`.

closed 2018-07-04 by insidioussid

nfeske, 2018-07-04:
Thanks a lot for your understanding. API-level decisions are often not clear-cut but sometimes rather blurry. So we had to come up with a few guiding principles.

The first is our ambition to optimize for low TCB complexity. If a change leads to a complexity reduction - especially of the base framework everyone has to rely on - we likely pursue it.

The second is orthogonality. A feature should exist only if it does not overlap with others. There are still some offenders in the current code base such as the old `Arg_string` utilities that could be covered with XML. So we will work towards removing the `Arg_string` over time.

The third is a rigid limit of scope. By declaring that Genode is _not_ an application-level API, we acknowledge to not even try catering application-level needs. The Genode API merely exists for the system to live. Everything outside this rigid scope should better be left out. It is a bit surprising that the bare API still allows us to build quite flexible applications and is quite joyful to use doing that. But that doesn't make it an application-level API.

When looking at the content accessor from this perspective, the bias against the inclusion become rather clear. (1) It does not make the base framework any simpler because there is no use case within the base framework. (2) It overlaps with the prevalent approach of encoding data as attribute values. (3) It serves application-level needs only (interfacing with 3rd-party tools).

BTW, I sympathize with you sentiment against pulling in complex 3rd-party libraries, especially if you have no use for the other features.