genodelabs/genode #2900: Cross-client I/O starvation at VFS server

ehmry, 2018-07-03:
It appears that a client processing I/O at least as fast as the VFS server will starve other clients. This is because the packet processing signal handler runs in a loop that only terminates when the client stops submitting packets or when the server can respond to packets faster than the client can submit them, which is unlikely.

ehmry, 2018-07-03:
Fixed on my `issue2900_fs_starvation` branch.

nfeske, 2018-07-03:
Is the test meant to be merged? It never terminates because the `run_genode_until` condition cannot be met:

```
child "test-fs_report" exited with exit value 0.*\n
```

May it be possible to have the test detect its success/failure? If so, it would be great to add it to the autopilot list. So we can keep an eye on the potential starvation effects on different kernels (e.g., depending on the order of signal delivery).

ehmry, 2018-07-04:
I fixed the test to terminate only if multiple clients are scheduled. I have a new series of commits on my `issue2900_fs_starvation` branch.

nfeske, 2018-07-04:
Thank you. I merged your new branch to staging.

ehmry, 2018-07-05:
I should mention that it would be easy to make a configuration knob that would adjust size of the packet processing batches for each client session, but for now I am content without one.

nfeske, 2018-07-06:
The fs_packet test fails on the following platforms:

```
qemu/fiasco_x86_32
qemu/foc_x86_32
qemu/foc_x86_64
qemu/hw_pbxa9
qemu/hw_x86_64
qemu/nova_x86_32
qemu/nova_x86_64
qemu/okl4_x86_32
qemu/pistachio_x86_32
qemu/sel4_x86_32
qemu/sel4_x86_64
x86/sel4_x86_64
```

ehmry, 2018-07-09:
Fixup for the test at 25b97ecdd796fd9f0db912a597096fa1e2f21373.

ehmry, 2018-07-11:
I have a fixup to disable the test for the combination of autopilot and QEMU at fad887c9388387c17f3afd778ee0b52981653baf.

commit ba29508b (2018-07-04): Process packets in batches at the VFS server

Process I/O packets in batches. If a batch is processed and there are
still packets pending, send a signal locally to the packet handler and
return to the entrypoint signal dispatcher. This prevents clients from
starving each other, which happens when a client continuously submits
packets at a faster rate than the server can process.

Fix #2900
commit b1b83f4d (2018-08-02): Process packets in batches at the VFS server

Process I/O packets in batches. If a batch is processed and there are
still packets pending, send a signal locally to the packet handler and
return to the entrypoint signal dispatcher. This prevents clients from
starving each other, which happens when a client continuously submits
packets at a faster rate than the server can process.

Fix #2900
closed 2018-08-08 by nfeske