genodelabs/genode #2902: sculpt: use vfs+cow for config fs

nfeske, 2018-07-04:
Sculpt uses the ram_fs component for the config file system. For Sculpt VC (visual composition), the config file system will host many small snippets of policy definitions, each hosted in a dedicated file. For this reason the ram_fs's support for pre-populating the file system with initial data from ROM modules becomes unsuitable. The use of the VFS server (with the initial content coming from a tar archive) overcomes this limitation. Thanks to the copy-on-write plugin, we can mutate the initial configurations.

ehmry, 2018-07-05:
I have a commit to bring the VFS to notification parity with the ram_fs at 7cc5c516066868754e97f3e312dd865843a2e1d2.

nfeske, 2018-07-06:
@ehmry this commit fixes the config-file update problems I had with sculpt after changing the config fs from ram_fs to vfs. Thanks a lot!

nfeske, 2018-08-08:
Sculpt uses the import-vfs-plugin for populating the config fs now. This mechanism turned out to be much simpler and predictable than the use of the cow plugin. (i.e., when removing a file from the config fs, the original file from the immutable cow-overlay does not re-appear)

closed 2018-08-08 by nfeske

ehmry, 2018-08-09:
Perhaps we should just remove the cow plugin? I no longer see a compelling use case for true copy-on-write that can't be solved as confidently as the with the import plugin.

nfeske, 2018-08-10:
I agree. We can still revive the idea once the time is right for it and we have a more precise understanding of the desired semantics (e.g., cow at a granularity of parts of files, removal of files, or recursion detection #2782). @ehmry would you like to provide a commit for the removal (code, recipe, adaptation of run scripts)?