genodelabs/genode #2903: Add vfs file size detection to ROM modules

ValiValpas, 2018-07-04:
Due to the fact that a ROM's size is given by its dataspace (and not the actual content), mounting a ROM into the vfs inevitably leads to an inflated file.

As ROMs typically contain strings. I suggest to let the `Vfs::Rom_file_system` detect the 0-termination to calculate the file size if `binary="no"` is added to the vfs config, e.g.:

`<vfs>
   <rom name="foobar" binary="no" />
</vfs>
`

Do you think this is a useful addition?

ValiValpas, 2018-07-04:
Great. What do you think about the referenced commit?

ValiValpas, 2018-07-04:
Nice.

What default do you prefer? If true, the previous behaviour remains unchanged if the binary attribute is not specified in the config.

nfeske, 2018-07-04:
The default should be true. I was a bit sloppy in my sketch. Thanks for pointing this out.

ValiValpas, 2018-07-04:
@nfeske Thanks for you quick feedback and your cool suggestions. I just pushed a fixup.

nfeske, 2018-07-04:
Very nice! I just added both commits as a single one to staging.

commit 6b6d6071 (2018-07-04): Add vfs file size detection to ROM modules

When mounting a ROM module with binary="no", the
vfs will detect the 0-termination to calculate the
file size instead of using the dataspace size.

Fixes #2903
commit c43ed44b (2018-08-02): Add vfs file size detection to ROM modules

When mounting a ROM module with binary="no", the
vfs will detect the 0-termination to calculate the
file size instead of using the dataspace size.

Fixes #2903
closed 2018-08-08 by nfeske