genodelabs/genode #2904: sculpt: use cached_fs_rom as depot_rom service

nfeske, 2018-07-04:
Sculpt currently employs the `fs_rom` server for accessing the depot content as ROM modules. This works but it has the following disadvantages:

* Since fs_rom reflects file changes as new ROM versions, it expects each accessed file to be potentially mutable and therefore registers watch handlers for the file and the directory tree hosting the file. This produces a lot of overhead in term of consumed capabilities but also RPCs during the ROM-session creation.
* In principle, each client may see a different version of a file depending on when the client updated the ROM session the last time. Therefore, fs_rom has to maintain a copy of the file in memory for each session. As each session loads the content from the file system, the content of unchanging files that are requested by many clients - like commonly used libraries - is redundantly accessed (consuming time) and stored (consuming memory).

Hence, the fs_rom server is not a good fit for the patterns how the depot is accessed. In fact, by design, the depot's content is immutable. So fs_rom's support for versioned ROMs remains unused while the overhead is being paid.

The new `cached_fs_rom` (#2760) is a much better alternative. So Sculpt should be changed to use this component as depot-ROM service. That said, for interactively developing depot content, the regular `fs_rom` has turned out to be a convenient tool. When using it as depot-ROM service, one can edit files within the depot and the changes become immediately into effect. E.g., when editing the backdrop's configuration, a running backdrop component immediately responds, which is pretty cool. To preserve this feature, I plan to add the following rule to Sculpt:

* By default all ROMs requested from the depot are obtained from the cached_fs_rom server.
* If the depot path refers to the user `local`, the ROMs are directed to the regular (dynamic) fs_rom server. This way, local modifications within /depot/local/ come immediately in effect.

Since the first case is the most common one, I'd call the cached_fs_rom "depot_rom" and the regular fs_rom "dynamic_depot_rom". I hope that this makes the distinction pretty intuitive.

ehmry, 2018-07-09:
Fixup at 975d3e31bd7b43fc1338545fbd75f8a62e77e311.

ehmry, 2018-07-09:
Still have issues with QEMU and Sculpt, will test tomorrow.

nfeske, 2018-07-10:
@ehmry The fixup solves my issue. Thank you for the super quick response. Added to staging.

commit 73ad1af3 (2018-07-09): sculpt: use cached_fs_rom as depot_rom

Fixes #2904
closed 2018-08-08 by nfeske