genodelabs/genode #2905: os: file system stress test

m-stein, 2018-07-04:


m-stein, 2018-07-04:
I had a look at the following candidates for a port:
* genio/doio
* bonnie++
* I read about fs tests with netperf but didn't found much
* iozone

I tend to give bonnie++ a try as source code and build system not seem complex to me, it's GPLv2 licensed, and it seems to be actively used and maintained.

ehmry, 2018-07-04:
I did this a year ago with IOzone, see #2905. From this experience I decided it would be better to record a trace of actual file-system activity and play it back without any pauses between operations, but didn't complete the trace or playback component.

chelmuth, 2018-07-05:
> I read about fs tests with netperf but didn't found much

I think you're referring to *netbench* which seems to be a commercial product. But there is (dbench)[https://www.systutorials.com/docs/linux/man/1-dbench/] which simulates a netbench run. I doubt this dusty *file-server* scenario is of much use for your stress test, so suggest to ignore it.

cnuke, 2018-07-05:
There is also [xfstests](https://git.kernel.org/pub/scm/fs/xfs/xfstests-dev.git).

nfeske, 2021-02-26:
The topic has become stale. There is no plan to pick it up in the foreseeable future. Hence, I'm closing it.

closed 2021-02-26 by nfeske