genodelabs/genode #2908: nic_router + foc + qemu: times out

m-stein, 2018-07-05:


m-stein, 2018-07-05:
The test works but needs like 4 min 30 sec to finish which is why it times out.
I made some measurements and found this:

**on x86_64 qemu nova**
total time: ~35 sec
heaviest thread:  by far "timer threaded_time_source" with ~12%
other threads: <5%
all threads: ~72%
kernel: ~28%

*on x86_64 qemu foc*
total time: ~270 sec
heaviest thread:  by far "timer ep" with ~10%
other threads: <2%
all threads: ~23%
kernel: ~77%

Thus, I assume that it is no problem with the userland code. I tried it without MP-support in foc, but then the test doesn't finish.

m-stein, 2018-07-05:
That said, I would vote for a higher test timeout for foc+qemu+x86_64

chelmuth, 2018-07-06:
It seems okay if only this specific platforms gets an extended timeout. It adds 5 minutes to our test run, though. Do you expect the test running on foc reveals additional issues or potential regressions with nic_router?

m-stein, 2018-07-06:
I cannot say if it's worth it. But yes, in the past we had enough issues not related to the nic_router at all  but triggered by this test.

m-stein, 2018-07-06:
At least, this measurement is another outcome of the test :)

nfeske, 2018-07-06:
I think that increasing the timeout for this specific conditions is the best way, or alternatively trimming the stress level of the test like we do in bomb.run, to not impede the delay on the overall tests.

In either way, the condition in the run script should best be accompanied with a comment that refers to this issue. So the knowledge about this anomaly does not get lost.

@m-stein The concrete numbers are indeed very interesting. Do you see a good way to preserve the mechanism used for obtaining them? It might become useful for cross correlating performance issues in the future.

m-stein, 2018-07-06:
Obtaining such numbers was done with a slightly modified version of the 'trace_logger' component (I made the output more readable for my specific task and added a calculation for total user utilization). A similar thing can be done with the 'top' component. But one must pay attention here because the "execution time" provided by the trace session has no unit, meaning that the unit depends on the kernel. Also the available subjects differ for the kernels. For instance, on nova the execution time is in microseconds and there are subjects "kernel idle"/"kernel cross" for each CPU core while on seL4 the execution time is in CPU cycles and on foc there are no subjects for kernel activities.

I would say that we should adapt the Trace session in a way that it always returns a 'Duration' value for the execution time and that it is clear whether a subject is a in kernel or not. With these two modifications, it would be much easier to produce cross-kernel comparable measurements.

m-stein, 2018-07-06:
By the way: For the reasons mentioned above, the meaning of the 'top' output currently differs for the different kernels. For instance, on nova, the percentage values refer to almost total system time while on foc they refer to time spent in userland only.

m-stein, 2018-07-06:
These two are a side effect of this issue:
a0d9034 */config.xsd: fix session-policy declarations
7b1e9ff base-foc: implement Platform_thread execution_time

m-stein, 2019-03-13:
This should fix the issue:
8c459d7ac2 run/nic_router foc x86_64: raise test timeout

Sometimes, the test still gets stuck, but this seems to be related to the socket/vfs connect problem.

commit 28913cfe (2019-03-14): run/nic_router foc x86_64: raise test timeout

Fixes #2908
closed 2019-03-19 by chelmuth