genodelabs/genode #2916: Strip "Name Service Switch" code from Libc

ehmry, 2018-07-12:
I don't know what exactly this NSS junk is but I think we don't want it. This would involve simplifying or replacing some of the upstream FreeBSD resolver code.

https://www.freebsd.org/cgi/man.cgi?query=nsdispatch&manpath=FreeBSD+11.2-RELEASE+and+Ports

ehmry, 2018-07-12:
For some background, my machine is building the libc library with a missing symbol for `_nsyylval` in  the NSS parsing subsystem, and I would rather remove any references to this global than fix the build.

The immediate problem with removing NSS is that it breaks resolving from `/etc/hosts`, because this file is accessed as a query to the NSS "hosts" database.

Must we now clarify what kind of resolver is baked into the libc? Should no resolver be included? Perhaps a DNS only resolver library or an NSS library should provide `getaddrinfo`?

chelmuth, 2018-07-12:
Regarding your original issue with `_nsyylval`, I see a recurring pattern of fixes required for issues with yacc/bison tool updates and are then always asking myself why we don't just archive one supported version of generated C source files. Are there any reasons contradicting this approach?

nfeske, 2021-02-26:
This issue has become stale.

closed 2021-02-26 by nfeske