genodelabs/genode #2919: Libc: improve passwd database support

ehmry, 2018-07-16:
The libc needs to make a better effort at providing some information about "users". The present dummy for ``getpwuid_r`` is broken because it fails to fill out a struct with pointers that application code will blindly dereference. Having better dummies here is important because for our libc `getenv("HOME")` cannot be expected to return anything.

This issue is related to using libssh for an ssh client.

ehmry, 2018-09-07:
Rebased against master on my `issue2919_getpwent` branch. I rewrote the Libc config accessor to simple return an `Xml_node`. I had a strange accessor before because I think I tried to make it thread safe, but the libc config ROM is never updated, so its not a concern of mine anymore.

chelmuth, 2018-09-10:
@ehmry fcc154f46efc433430460e8eb47e0d45f94c1e1f breaks `make run/stdcxx KERNEL=linux` (please consider to test together with fixup 69b771daac5bffc6012ad4cd21cb6951f3ce8e22)

chelmuth, 2018-09-10:
`make run/vfs_import KERNEL=linux` also chokes on the commit mentioned above.

ehmry, 2018-09-10:
I have a fix for stdcxx at 15be3b53fb344bde9faaa9f71a1801a34503db5a, the problem with noux I suspect may be `fsync(0);`.

nfeske, 2018-09-11:
The noux patch does not look right to me. The lifetime of the I/O channels is managed via reference-counting shared pointers. So the correct way would be dropping the pointers by assigning invalid (default-constructed) 'Shared_pointer` values to the `_channel_0` ... `_channel_2` variables. To check that this triggers the desired destruction, it might good to have a look at the `count()` value of the pointer, to see if the code removes the last reference.

ehmry, 2018-09-11:
@nfeske yes, assignment is suffcient to close these channels, a better patch at c93cba4e28507c959d1a5e05211e65faa476dd55.

commit b3b48bdc (2018-09-07): Libc: single-user getpwent implementation

Implement the passwd database subroutines with single-user database.
This database is populated with a <passwd/> sub-node of the libc
configuration node. All fields of the "passwd" struct may be specified
with reasonable defaults provided for a "root" user. This allows a
libc-based component to spoof user information for the sake of porting
existing Unix software.

A test is provided at run/libc_getpwent.

Fix #2919
commit 29d946aa (2018-09-13): Libc: single-user getpwent implementation

Implement the passwd database subroutines with single-user database.
This database is populated with a <passwd/> sub-node of the libc
configuration node. All fields of the "passwd" struct may be specified
with reasonable defaults provided for a "root" user. This allows a
libc-based component to spoof user information for the sake of porting
existing Unix software.

A test is provided at run/libc_getpwent.

Fix #2919