genodelabs/genode #2920: libc: 'connect()' erroneously reports success when used with VFS server

cproc, 2018-07-17:
The test case in commit [1] tries to connect to 'genode.org:81', which should fail, but the 'connect()' call does not fail when using the VFS server.

good case: vfs_lxip plugin used locally:

```
[init -> test-libc_connect] socket() returned 4
[init -> test-libc_connect] connect() returned -1
```

bad case: vfs_lxip plugin used via VFS server:

```
[init -> test-libc_connect] socket() returned 4
[init -> vfs] Error: partial write detected 0 vs 17
[init -> test-libc_connect] Error: packet operation=1 failed
[init -> test-libc_connect] connect() returned 0
```

[1] https://github.com/cproc/genode/commit/987132aed38dacc6f1b15b76cf467f7d4b847324

ehmry, 2018-07-23:
As I recall there was a solution to this by calling sync after writing to the connect, so either that is not implemented or it doesn't work over the FS session.

ehmry, 2018-07-23:
Fixed on my `issue2920_libc_connect` branch.

ehmry, 2018-08-07:
LwIP fix at c3fac5d746ffe43836f33408e50974c662052994.

commit 541b4ccf (2018-08-02): LxIP: propagate write errors using SYNC_ERR_INVALID

The libc sockets implementation already syncs socket control files after
writes, so sync errors will induce failures for operations such as
"connect".

Fix #2920
closed 2018-08-08 by nfeske