genodelabs/genode #2922: depot: create api recipe for gems/include/depot

trimpim, 2018-07-18:
I'm working on a component that needs to handle depot paths. Without this recipe I can't create a binary package for the component.

nfeske, 2018-08-06:
On merely technical grounds, adding an API for a single header looks quite bureaucratic to me.

Right now, each source archive that needs gems/include/depot includes a copy of the header (like https://github.com/genodelabs/genode/blob/master/repos/gems/recipes/src/depot_deploy/content.mk), which is - in my opinion - simple and flexible.

An API archive is useful when it the content is very large (so that copying the content into each src archive becomes prohibitive), or when it serves as a front-end of an ABI (like for ported libraries that contain an accompanied symbols file), or when it is very prominently used (so that shipping copies of headers in src archives becomes a nuisance). This is not the case for include/gems/depot.

However, maybe you want to keep the content of a custom src archive free from Genode headers to simplify the archive's license information? In this case, wrapping up the needed headers into a separate archive makes sense.

In other words, if it makes your life easier, I'd merge your commit. But should the copy-into-src-archive approach work well for you, I'd rather not merge it.

trimpim, 2018-08-06:
Spam detection software, running on the system "obelix.duss.local",
has identified this incoming email as possible spam.  The original
message has been attached to this so you can view it or label
similar future email.  If you have any questions, see
root@localhost for details.

Content preview:  Hello Norman I hope you had a pleasant holiday. Quoting Norman
   Feske <notifications@github.com>: 

Content analysis details:   (7.1 points, 5.5 required)

 pts rule name              description
---- ---------------------- --------------------------------------------------
 2.9 HELO_DYNAMIC_SPLIT_IP  Relay HELO'd using suspicious hostname (Split
                            IP)
 0.2 CK_HELO_GENERIC        Relay used name indicative of a Dynamic Pool or
                            Generic rPTR
 3.6 RCVD_IN_PBL            RBL: Received via a relay in Spamhaus PBL
                            [178.194.91.161 listed in zen.spamhaus.org]
 0.0 T_SPF_HELO_TEMPERROR   SPF: test of HELO record failed (temperror)
 0.4 RDNS_DYNAMIC           Delivered to internal network by host with
                            dynamic-looking rDNS

nfeske, 2018-08-06:
Hi @trimpim, yes I had a fantastic time in Norway. ;-) Now, feeling refreshed, I am ready to work towards the upcoming Genode release.

The rest of your comment remains a mystery. I guess something went wrong with submitting your comment via email?

trimpim, 2018-08-06:
Sorry, for the crumbled  reply, somehow my webmail doesn't like our office ;-).

>  On merely technical grounds, adding an API for a single header looks quite bureaucratic to me.

I agree with you on this.

> Right now, each source archive that needs gems/include/depot includes a copy of the header (like https://github.com/genodelabs/genode/blob/master/repos/gems/recipes/src/depot_deploy/content.mk), which is - in my opinion - simple and flexible.
I did forget this possibility.

I have tried to use this solution. Unfortunately this doesn't seem to work for me.
- Firstly I needed to add some ../../ magic that make finds the path (the user is in an other repo that is included via a symbolic link).
- Secondly I got the following error after I had this sorted out:
.../src/app/slice_layouter/main.cc:11:27: fatal error: depot/archive.h: No such file or directory
 #include <depot/archive.h>

nfeske, 2018-08-07:
Sorry that my example was a bit misleading. I did not mean to use the mirror_from_rep_dir macro but just the general approach of copying unpackaged headers into a src archive. An example better suited as a blue print is https://github.com/genodelabs/genode/blob/master/repos/gems/recipes/src/floating_window_layouter/content.mk Here, the window layouter includes some headers from the include/decorator directory, hosted in the os repository. As you can see, you can use $(GENODE_DIR) in content.mk files to refer to the top-level genode directory. So you can grab archive content from anywhere in the source tree.

trimpim, 2018-08-10:
Sorry for the delay in my answer, but I was working on an other problem.
You do not need to merge the commit. I was able to solve my problem with your description.

closed 2018-08-10 by trimpim