genodelabs/genode #2923: Terminal: VT100 compatiblity

ehmry, 2018-07-19:
The Terminal component is not effectively at parity with the capabilities of the VT100 or GNU Screen.

ehmry, 2018-08-07:
Please conside 69d3b7ea2eee4900851554a4b86424d885dd5700 for staging, this allows the terminal to survive unhandled escape sequences.

nfeske, 2018-08-08:
This is a very profound update. I like how your additions neatly follow the lines of the existing code, which makes to possible for me to (roughly) follow the changes. I especially like how your patch fills a lot of blank spots (_missing) with sensible code. If it makes the terminal more robust against still still unhanded or invalid escape sequences, all the better.

ehmry, 2018-08-08:
@nfeske fixup to turn strict checks on at cba6b55a83a8e286d113ac4966e912ebf47ab757.

nfeske, 2018-08-08:
Thanks. I squashed both commits onto staging.

commit 2041f957 (2018-08-08): Terminal: improved handling of escape sequences

Add additional parsing modes to the sequence decoder to detect and
discard unhandled sequences for ECMA-48, DEC private, and Xterm.

Add new behavior for cursor movement, cursor hiding, character deletion,
and line-wrapping.

Fix #2923