genodelabs/genode #2929: hw/x86: add SMP support

alex-ab, 2018-08-03:


nfeske, 2018-08-08:
@alex-ab the "fixed" tag suggests that your commit series is ready for staging. I would like to merge but wonder, did @skalk or @m-stein already have a look at the changes?

m-stein, 2018-08-08:
I had just some minor annotations. Nice commit!

alex-ab, 2018-08-09:
Thanks for the review. I incorporated your comments and rebased to current staging on this [branch](https://github.com/alex-ab/genode/tree/issue_2929_staging)

m-stein, 2018-08-09:
Looks great! Thanks!

skalk, 2018-08-13:
@alex-ab I'm fine with it. Thank you for your efforts!

alex-ab, 2018-08-16:
@nfeske: please consider for staging https://github.com/alex-ab/genode/commits/issue_2929_staging

nfeske, 2018-08-16:
Thanks for the reminder. Merged right away. ;-)

alex-ab, 2018-08-17:
@nfeske, @chelmuth: please consider e1237e690d9a81e5cbd2f0ba16e84fbde2373f68  for staging to make muen/hw happy

alex-ab, 2018-08-17:
Additionally, in contrib/muen-a4ce6a1f3bc1209e4c9f8ecf163d7af354c188af/src/kernel/muen/policy/xml/vcpu_subject_base_hw.xml respectively build/muen/kernel/muen/policy/xml/vcpu_subject_base_hw.xml the start rip must be changed from ```16#0020_0028``` to ```16#0020_0078```

@reet-, @Kensan: may you please provide a updated muen branch with the xml change ?

Kensan, 2018-08-17:
@alex-ab Could you carry a local patch, analogous to how @chelmuth did in 7f2087e105f14460c30da128b3b9e77de5dd50a8, for now until we get around to testing Genode staging with the current Muen devel branch?

We will submit a Muen port update as soon as we get around to it, but this may take a bit of time.

alex-ab, 2018-08-17:
Ok. @chelmuth, @nfeske: please add 505173063d96cb77ef60b4779f12ae1436fee194 to staging

alex-ab, 2018-08-17:
I'm not sure. Even if you move the 16bit code down (which should work), I fear that the factored out paging setup code (as macro) may cause trouble. Please try yourself, my time is over ;-)

commit 565e656d (2018-08-16): hw/x86: enable SMP support

Fixes #2929