genodelabs/genode #2930: fb_sdl.cc:23:21: fatal error: SDL/SDL.h: No such file or directory

HenrikBach1, 2018-08-04:
Hi,

I've build the tool-chain from source and now I'm trying to build the `demo` program from the source. However, the build fails due to it cannot find a SDL.h include file:

```dos
#
# Create tool-chain
#
$CMD ${wrk_dir}/./tool/create_builddir x86_64
...
$CMD ${wrk_dir}/./tool/tool_chain x86
...
installing tool chain to '/usr/local/genode-gcc'...
...
$CMD /bin/bash -c "cd ${wrk_dir}; KERNEL=linux make run/demo"
++ docker exec --workdir / genode-ubuntu-16.04 /bin/bash -c 'cd /projects/genode/build/x86_64; KERNEL=linux make run/demo'
including /projects/genode/tool/run/power_on/linux
including /projects/genode/tool/run/log/linux
including /projects/genode/tool/run/boot_dir/linux
including /projects/genode/repos/os/run/demo.run
building targets:  core init drivers/timer server/nitpicker app/pointer app/status_bar app/global_keys_handler app/nit_focus server/liquid_framebuffer app/launchpad app/scout test/nitpicker server/nitlog drivers/framebuffer drivers/input server/report_rom server/rom_filter server/report_rom drivers/acpi
spawn make core init drivers/timer server/nitpicker app/pointer app/status_bar app/global_keys_handler app/nit_focus server/liquid_framebuffer app/launchpad app/scout test/nitpicker server/nitlog drivers/framebuffer drivers/input server/report_rom server/rom_filter server/report_rom drivers/acpi
make[1]: Entering directory '/projects/genode/build/x86_64'
...
  Program drivers/framebuffer/spec/sdl/fb_sdl
    COMPILE  fb_sdl.o
/projects/genode/repos/os/src/drivers/framebuffer/spec/sdl/fb_sdl.cc:23:21: fatal error: SDL/SDL.h: No such file or directory
 #include <SDL/SDL.h>
                     ^
compilation terminated.
/projects/genode/repos/base/mk/generic.mk:56: recipe for target 'fb_sdl.o' failed
make[3]: *** [fb_sdl.o] Error 1
var/libdeps:486: recipe for target 'fb_sdl.prg' failed
make[2]: *** [fb_sdl.prg] Error 2
make[2]: *** Waiting for unfinished jobs....
```

I've installed the `libsdl2-dev` library without any complaints from building the tool-chain.

What did I do wrong? 

Help is appreciated. 

Thank you.

/Henrik

cnuke, 2018-08-04:
Please install `libsdl1.2-dev` because the `fb_sdl` component still uses the old SDL version.

HenrikBach1, 2018-08-06:
Yes, it compiles without any complaints. Issue is closed.

HenrikBach1, 2018-08-09:
Hi,

Yes it compiled, but the shell in the Demo didn't show up in my virtual machine. Therefore, I setup above code and tools on physical hardware.

However, the compilation/linking stops at `cpu_session_support.o`:

```bash
...
  Program core/linux/core-linux
    COMPILE  cpu_session_component.o
    COMPILE  version.o
    LINK     core-linux
cpu_session_support.o: file not recognized: File truncated
collect2: error: ld returned 1 exit status
/projects/genode/repos/base/mk/prg.mk:193: recipe for target 'core-linux' failed
make[3]: *** [core-linux] Error 1
var/libdeps:378: recipe for target 'core-linux.prg' failed
make[2]: *** [core-linux.prg] Error 2
Makefile:275: recipe for target 'gen_deps_and_build_targets' failed
make[1]: *** [gen_deps_and_build_targets] Error 2
make[1]: Leaving directory '/projects/genode/build/x86_64'
Error: Genode build failed
Makefile:312: recipe for target 'run/demo' failed
make: *** [run/demo] Error 252
```

What am I doing wrong?

Thank you.

/Henrik

chelmuth, 2018-08-09:
Please try
```
rm core/linux/cpu_session_support.o; make core KERNEL=linux
```

HenrikBach1, 2018-08-09:
Somehow, every *.o file was corrupted. When all *.o files were deleted and rebuild, everything worked like a charm.

/Henrik

closed 2018-08-09 by HenrikBach1