genodelabs/genode #2935: acpica port: update to version 2018-08-10

m-stein, 2018-08-20:


nfeske, 2018-08-29:
On my x260, the new version of acpica gives the following error:

```
[runtime -> acpica] Error: AcpiOsReadPciConfiguration unknown device - segment=0 bdf=00:1f.4  reg=0xe width=0x8
[runtime -> acpica] Error: AcpiOsDeleteSemaphore:128 called - dead
```

chelmuth, 2018-08-29:
From the background of the error above it may be worthwhile to investigate ACPI_SINGLE_THREADED, which has a pretty simple implementation in osunixxf.c for semaphores.

m-stein, 2018-08-29:
I'll try this, thanks for the hint.

alex-ab, 2018-08-29:
Seems reasonable, ack.

alex-ab, 2018-08-31:
@chelmuth: the AcpiOsDeleteSemaphore fix did not get to the release somehow

chelmuth, 2018-09-03:
This is because I did not commit it. I just created the patch for discussion, sorry for the confusion.

alex-ab, 2018-09-03:
Please add d73ee020c06b016b83a7dd77e1db1343d53c55e5 to staging, which fixes the reported regression and another one happening on "poweroff". 
@nfeske: Please consider 9806e50f142c83274f7960230c7a12f0d63e0dcb (or similiar) for sculpt, to enable (ease) the poweroff/reset support

commit 7cfa2331 (2018-08-21): acpica port: update to version 2018-08-10

Fixes #2935
commit a64c29c6 (2018-09-03): acpica: fix regressions (poweroff and semaphore)

introduced by acpica update to version 2018-08-10.

Fixes #2935
commit 39ad6359 (2018-09-03): acpica: fix regressions (poweroff and semaphore)

introduced by acpica update to version 2018-08-10.

Fixes #2935
commit 50dcf40f (2018-09-05): acpica: fix regressions (poweroff and semaphore)

introduced by acpica update to version 2018-08-10.

Fixes #2935
closed 2018-09-05 by nfeske