genodelabs/genode #2937: sculpt: config file modifications remain unnoticed under some conditions

nfeske, 2018-08-21:
After the sequence of the following commands in the inspect window, the sculpt manager stops responding to any further file modification:

```
cp /config/deploy /config/deploy.bak
mv /config/deploy.bak /config/deploy
```

The same effect occurs when applying the commands to other config files such as /config/nitpicker.

It looks like the `mv` command produces an access pattern that confuses `fs_rom` but that's just a guess. @ehmry Would you like to give it a look?

ehmry, 2018-08-21:
This might be fixed by 8252f17969add90fc91d527af71f304265fd2943, but I'm not able to readily test it right now on this laptop.

nfeske, 2018-08-22:
Your patch fixes the issue. Merged to staging. Thank you very much!

commit b84c891c (2018-08-22): VFS: Send notifications on RAM file rename

Fix #2937