genodelabs/genode #2941: Large Nitpicker dimensions may trigger blocking buffer upgrade loop

ehmry, 2018-08-24:
When a Nitpicker client that is packaged with low memory requirements is launched in an environment with buffers that are larger and mode expensive than expected the client may enter into a n upgrade loop that stalls waiting for quota from parent.

nfeske, 2018-08-24:
Does the client trigger a resource request? If yes, the current behavior is actually intended. The parent who receives the resource request can decide to continue the operation of the client by increasing the client's quota accordingly. I think aborting is worse than escalating the resource-scarcity problem to the parent.

I guess the issue is annoying right now because the client is hosted behind a subinit, which happens to never respond to the resource request. It would be good to enhance init with an option such that resource requests by a child result in a resource request of init itself. If that would be the case, the end user may interactively respond to such a situation by upgrading the resources of init, which, in turn, would forward these resources to the blocked child, unblocking it. The details and configuration interface of such dynamic resource handling are still not completely clear though.

nfeske, 2018-09-05:
@ehmry how we should proceed regarding this issue?

ehmry, 2018-09-05:
I will close this for now and open it again if it becomes a more pressing concern.

closed 2018-09-05 by ehmry