genodelabs/genode #2945: Solo5 support

ehmry, 2018-08-27:
I have made some research into the platform layer of [MirageOS](https://mirage.io/) and found two means of running Mirage on Genode, the first is to port the OCaml runtime to Genode and write OCaml libraries to interface with Genode services. The second method would be to simply port the [Solo5](https://github.com/Solo5/solo5) middleware to Genode, and Mirage would continue to use the OCaml libraries for interfacing with Solo5.

Having made a cursory inspection of Solo5, I am unconvinced that any "virtualization" is necessary for Genode "deployment". The sandboxing inherent to the Genode framework should provide the same level of isolation for a "unikernel" as would the Linux flavor-of-the-month virtualization. To test this theory I have reimplemented Solo5 from scratch to meet the high-level interface in the `solo5.h` header and have successfully tested it against the tests provided with Solo5. I have not tested this Solo5 implementation with any unikernels and expect some problems with the current memory and ELF image layout.

My earlier version of Solo5 is at d531f124f326cfa32a875a09f518bacea1d2cca0. I would defer work with Mirage and other unikernels until a Genode SDK is provided for integration with the OPAM package manager. OPAM is used by the Mirage build tooling to build and link Mirage to Solo5.

ehmry, 2018-08-28:
Depends on #2949.

ehmry, 2018-11-07:
Fixed at b6e811f151b21f1a3fcd9d31935b8ed8cd644700, this patch adds rules for building the C++ implementation in the upstream repo, adds a test to autopilot, and adds *api* and *src* recipes.

ehmry, 2018-11-07:
@chelmuth, @nfeske the next Mirage release is this week, can I get this patch on staging or master quickly?

https://github.com/mirage/mirage/issues/939

nfeske, 2018-11-08:
It's my pleasure! :-) Merged to staging.

chelmuth, 2018-11-12:
Currently, we have build issues on 32-bit. Is this supported by solo5?
```
[2018-11-12 04:00:39] /data/genode/contrib/solo5-38981f1a735dfaa4bffc241db5d3b31e2b47163c/src/lib/solo5/bindings/genode/bindings.cc:443:36: error: narrowing conversion of ‘(deadline_us - now_us)’ from ‘solo5_time_t {aka long long unsigned int}’ to ‘long unsigned int’ inside { } [-Werror=narrowing]
[2018-11-12 04:00:39]    Genode::Microseconds{deadline_us - now_us});
```

Also, I'm wondering if it makes sense to build solo5 on ARM, @ehmry ?

ehmry, 2018-11-12:
@chelmuth Solo5 supports x86_64 and ARM64, so the best thing to do for now is to require the `64bit` spec, fixup at 2c0f30a8b58f3988954d881c16212e1325865915.

commit cfbab5f7 (2018-11-08): Native Solo5 bindings

A shared library implementation of the unikernel middleware.

https://github.com/Solo5/solo5

Fix #2945
closed 2018-11-16 by chelmuth