genodelabs/genode #2947: libc http client test

cproc, 2018-08-27:
Commit [1] implements a simple http client test which downloads the genode.org website in a loop using the libc socket API. The test can be used single-threaded or multi-threaded (bool value in the application) and with vfs_lwip or vfs_lxip locally or in the VFS server (individual run scripts).

The test is not ready for staging, but might be helpful to debug the problems that are triggered by it. 

The current state with VFS server fixup commit [2] applied is:

libc_http_client_vfs_lwip_local.run (single thread): read() returns 0
libc_http_client_vfs_lxip_local.run (single thread): hangs (see [3])
libc_http_client_vfs_lwip_server.run (single thread): ok, but with error messages
libc_http_client_vfs_lxip_server.run (single thread): ok

libc_http_client_vfs_lwip_local.run (two threads): read() returns 0
libc_http_client_vfs_lxip_local.run (two threads): hangs (see [3])
libc_http_client_vfs_lwip_server.run (two threads): hangs (see [4])
libc_http_client_vfs_lxip_server.run (two threads): hangs (see [4])

The test can be compiled for Linux (g++ -std=c++11 -pthread main.cc) and works ok on Linux, both single-threaded and multi-threaded.

[1] https://github.com/cproc/genode/commit/ca190b297396c386813e537be7cd943b0057a196
[2] https://github.com/genodelabs/genode/commit/9b390b02273c904948680f27543d4794176458ea
[3] could be the known deadlock problem caused by `wait_and_dispatch_one_io_signal()` in lxip
[4] works when calling `_process_packets()` instead of `_process_backlog()` in commit [2]

ehmry, 2018-08-28:
There is a bug in the LwIP VFS plugin, the chained buffers used by Lwip use sizes expressed in sixteen bits, and under conditions such as a read of 1<<16 the higher bits are lost and the plugin performs a zero length read, and the application interprets this as a closed connection. Fixed at edf81385da79d484f50e0141ae8ac708048bc8d0.

ehmry, 2018-08-28:
I recommend for inclusion in 18.09. The FS client already truncates reads to fit in the packet buffer, so I don't expect any side effects.

cproc, 2018-08-31:
The state of the test on the current staging branch is:

libc_http_client_vfs_lwip_local.run (single thread): ok
libc_http_client_vfs_lxip_local.run (single thread): hangs
libc_http_client_vfs_lwip_server.run (single thread): ok, but with error messages
libc_http_client_vfs_lxip_server.run (single thread): ok

libc_http_client_vfs_lwip_local.run (two threads): various error symptoms
libc_http_client_vfs_lxip_local.run (two threads): hangs
libc_http_client_vfs_lwip_server.run (two threads): hangs
libc_http_client_vfs_lxip_server.run (two threads): hangs

chelmuth, 2019-12-20:
May we add these test to our test bench?

nfeske, 2021-02-26:
@cproc How should we go about this issue?

cproc, 2021-02-26:
@nfeske: if I remember correctly, the last plan was to enhance the 'tcp-bulk' test with the multi-threading feature of this test. So we could close this issue for now and open a new one when somebody is going to work on the 'tcp_bulk' test.

nfeske, 2021-02-26:
Thank you @cproc for commenting.

commit b5d38284 (2018-08-29): LwIP: clamp reads to a value that can be expressed in 16 bits

There is a bug in the LwIP VFS plugin, the chained buffers used by Lwip
use sizes expressed in sixteen bits, and under conditions such as a read
of 1<<16 the higher bits are lost and the plugin performs a zero length
read, and the application interprets this as a closed connection.

Fix #2947
commit 6f8e8a6e (2018-08-30): LwIP: clamp reads to a value that can be expressed in 16 bits

There is a bug in the LwIP VFS plugin, the chained buffers used by Lwip
use sizes expressed in sixteen bits, and under conditions such as a read
of 1<<16 the higher bits are lost and the plugin performs a zero length
read, and the application interprets this as a closed connection.

Fix #2947
closed 2021-02-26 by nfeske