genodelabs/genode #2948: Genode SDK for third-party tooling

ehmry, 2018-08-28:
The current recommendation is that Genode components only be built using the build system supplied in the base code repository. This is natural because for C and C++ development each project must provide its own code and build management, and for these languages I expect this to recommendation to persist indefinitely. In other languages, in particular Nim and OCaml, there is a strong expectation that all code to be published should be complied and linked by official packaging tools.

Building Nim components with the Make build system has always been problematic because the Nim build process requires three steps, Nim translation, C++ translation, and linking. This steps should be performed by the Nim compiler but in the case of the main repositories the last two steps are performed by Make with our tools, but external Nim dependencies must be managed manually. For the world repository the Nimble package manager manages depenencies and performs the first two steps, but the next Nim release will introduce a global build cache for objects that will make linking even more tedious.

The OCaml relies on the OPAM package manager for managing libraries and even the OCaml compiler version. OPAM packages are also used for building object files in other languages, this is the case for the Solo5 environment of the Mirage unikernels. Genode simply cannot be integrated into the Mirage tooling without an OPAM package for platform support.

I believe it would be possible to build Genode components with Nimble and OPAM if an SDK was made available that contained the toolchain, headers, and ABI stubs. Both package managers support dependencies on packages provided by Linux distributions, so ideally the SDK would be served from a Debian package repository. Adding an external package repository would not be automatic, but once that step is manually completed the language package managers would be able to print helpful errors about missing SDKs and version mismatches.

I roughly imagine the format of the SDK as follows:
```
/opt
  genode-sdk -> genode-sdk-18.11 # Symlink to active SDK
  /genode-sdk-18.11
    /gcc     # Patched toolcain
    /include # Merged repo headers, no kernel headers
      /x86_64 # Merged repo headers for a hardware spec
    ...
    /lib     # Stub ABI libraries
      /x86_64 # Libraries for a hardware spec
      ...
```

The SDK would be built by a makefile in the Genode code repository. Generally native headers and libraries would be included along with POSIX and Stdc++, as those are considered basic langauge runtimes. SDL, Qt, and other ported software would be ommited.

Does this sound reasonable? I consider this a requirement for OCaml support and the Nim-0.19 release (but no idea when that is).

nfeske, 2018-08-28:
That sounds very very good!

Do you plan to assemble the sdk content directly from depot archives (api archives + matching lib binary archives)? This would be very handy for extending the SDK with additional hand-selected libraries (such as ncurses) in the future.

chelmuth, 2018-08-28:
This sounds as a perfect first step to a more convenient integration of Qt5 development tools with Genode. So, I'd add the libc/qt5 by the means @nfeske describes too.

ehmry, 2018-08-28:
In my current iteration (27f0ddfcce582b587dcda6824138ecfc2757ba82) I am simply copying files from `$(GENODE_DIR)` into an `/include` and generating the ABI stubs on the fly into `/lib`. That makes it easy to pass other tools `-I/opt/genode-sdk-x86_64/include` but perhaps keeping the granularity of depot packages would be good. The current depot revisions could be copied into the SDK as unversioned directories such as `/api/libc`, `/api/base`, `/api/os` and leave it to the integrator choose what they need.

nfeske, 2018-08-28:
Yes, this sounds pragmatic (good).

It would be useful to preserve the information about the used archive versions in the SDK though. For example, with this information, one can create depot archives for the cross-compiled applications, which then refer to actually existing genodelabs depot content. (I assume that ultimately, one wants to get out a deployable pkg more or less automatically)

ehmry, 2018-08-28:
The files in the SDK should be read-only, so I think the spec directories be merged into simple mutually exclusive platforms like x86_64, x86_32, riscv_64 and allow for some duplication.

ehmry, 2018-09-03:
I have a simple patch to begin preparations, f980c80f75291809750c3dc2811653ed9d22d787 moves the platform-specific libc headers so that generic SPECS rules can be applied to the libc API package.

jklmnn, 2018-09-04:
This also sounds like a good approach to use gprbuild to build Ada on Genode.

nfeske, 2018-11-22:
@cnuke how do you feel about merging @ehmry's version of the SDK? As far as I remember, you had proposed some modifications (e.g., library-specific sub directories).

I wonder, how should we go about the SDK and #2949 regarding the upcoming release? Don't worry, I don't want to rush it.

cnuke, 2018-11-22:
@nfeske I'm fine with merging @ehmry's current version, it includes the proposed changes AFAICT.

ehmry, 2018-11-23:
@nfeske see df67994fb00a20daccc4173c3bcf66541bdfb7f4.

nfeske, 2018-11-23:
Very good. I merged the SDK patch to staging now.

ehmry, 2018-11-26:
I have a fixup at 6a06f68203efd4d9923d3859beb54729be3e5592 that fixes the usage message and adds ldso-startup.lib.so to the libraries. Adding this static library is actually quite easy, and is necessary to build libraries with globals (libretro cores).

nfeske, 2018-11-26:
Merged to staging.

commit 918281b0 (2018-11-29): Genode SDK

Makefile for generating a Genode SDK.

Fix #2948
closed 2018-11-29 by chelmuth