genodelabs/genode #2950: nova: rm_nested test hangs sporadically

alex-ab, 2018-08-29:
```
[init -> test-rm_nested] write to 0x26000
[init -> test-rm_nested] region-map state is WRITE_FAULT, pf_addr=0x6000
[init -> test-rm_nested] allocate dataspace and attach it to sub region map
[init -> test-rm_nested] returning from handle_fault
page fault, pd='init -> test-rm_nested' thread='ep' cpu=0 ip=0x1000458 address=0x26000 stack pointer=0xa04fef18 qualifiers=0x6 irUWp reason=1
Error: Test execution timed out
```

alex-ab, 2018-08-29:
The reason is a race between pager thread, ep thread and a (core external) region manager thread.
During the generic [pager code call](https://github.com/genodelabs/genode/blob/master/repos/base-nova/src/core/pager.cc#L125) a signal to the external region manager thread may be issued. If the region manager thread manages to answer the fault quick enough and calls into core via the [ep thread](https://github.com/genodelabs/genode/blob/master/repos/base-nova/src/core/pager.cc#L407), the wakeup of the thread is dropped. In this case, the pager thread sets the faulting thread too late on a [blocked state](https://github.com/genodelabs/genode/blob/master/repos/base-nova/src/core/pager.cc#L158).

nfeske, 2018-09-05:
Merged to staging.

commit bc9f0fdc (2018-09-05): nova: lock pager object state during fault lookup

Fixes #2950