genodelabs/genode #2951: rtc_drv: wrong binary taken for Linux or Microkernel on x86

alex-ab, 2018-08-29:
When using the same build directory, e.g. build/x86_64, for KERNEL=linux and KERNEL=some_micro_kernel, the rtc_drv will be not rebuild in all situations. This leads to the issue that on Linux e.g. the rtc_drv tries to open IO_PORT session, which ultimately fails.

chelmuth, 2018-08-29:
Prior discussions lead us to the follow procedure.
1. Rename the drivers appropriately, e.g. *x86_rtc_drv* and *linux_rtc_drv*.
2. Use depot packages containing appropriate drivers for the selected platform in run scripts.

chelmuth, 2018-08-29:
This issue is closely related to #2190 and #2403.

alex-ab, 2018-08-29:
Does 0c6cca72faf5f53245f94e14cc78f72fdd7c5dcf qualify ?

chelmuth, 2018-08-29:
That's unfortunately the wrong direction as the *binary_name* run-script mechanism should not be scaled to more drivers.

jklmnn, 2018-10-29:
> Prior discussions lead us to the follow procedure.
> 
>     1. Rename the drivers appropriately, e.g. _x86_rtc_drv_ and _linux_rtc_drv_.
> 
>     2. Use depot packages containing appropriate drivers for the selected platform in run scripts.

I'd prefer the first variant since it allows to specifically use use the `x86_rtc_drv` with #2829. Otherwise one needs to patch out the Linux specific parts.

chelmuth, 2018-10-29:
I'm afraid you got me wrong, 1) and 2) are steps of one procedure. Unfortunately, my patch is only half-baked and still needs some fixes.

jklmnn, 2018-10-29:
Oh, well that would still solve the problem since I would be able to select the correct driver manually without needing to patch it.

chelmuth, 2019-06-12:
@skalk your work to release 19.05 fixes this issue, right?

skalk, 2019-06-12:
Now, Linux driver is named `linux_rtc_drv`, but the x86 native driver is still named `rtc_drv`. So, yes the drivers are named unambigously and can be build independently in the same build directory.

skalk, 2019-06-12:
That being said, there is no package refering the rtc_drv component.

chelmuth, 2019-06-13:
So, we should finish the renaming for *x86_rtc_drv* too.

skalk, 2019-06-13:
I did not do this in the past to not break run-scripts outside without a need, now that there are no name clashes anymore. BTW. the same applies to, e.g., usb_drv.

nfeske, 2019-06-13:
I'm with Stefan here. The name rtc_drv and usb_drv is fine with the background that the x86 versions are the most commonly used ones (used across all PCs including AMD and Intel).

chelmuth, 2019-06-18:
Under these circumstances... I close the issue as fixed.

closed 2019-06-18 by chelmuth