genodelabs/genode #2954: VFS import cannot write large files thru File_system client

ehmry, 2018-08-29:
The VFS import pluging does not handle the `Vfs::File_io_service::Insufficient_buffer` exception thrown by the `File_system` client when the packet buffer congests with write packets. I would prefer to replace this exception with an error value, but I'm not familiar with the history of the exception.

ehmry, 2018-08-29:
Fix at 4a81139a02a9af8a52a2ac77be0806f514def5c5.

ehmry, 2018-08-30:
Disregard this fix, there is also a problem with selecting write buffer sizes here.

nfeske, 2018-08-30:
BTW, I was wondering why your patch blocks for exactly one signal in the exception case.

ehmry, 2018-08-30:
There is a loop, so its a block and retry.

ehmry, 2018-09-03:
I have a rather massive patch at 7ad04e63969f0ebe23a3eb3fac34008bd9452bcc as discussed offline. This cleans up the errors and exceptions, as well as some of the VFS server packet handling.

ehmry, 2018-09-11:
Rebased at 0fe3c6747962f2abc748c6621bb7706584330cd3 with the addition of a `WRITE_OVERSIZED` error to fix #2971.

chelmuth, 2018-09-13:
With commit 5d7b8321e6e082d980d4d41689639fad67cf3a46 I get

```
repos/libports/src/lib/libc/vfs_plugin.cc:463:9: warning: enumeration value ‘WRITE_OVERSIZED’ not handled in switch [-Wswitch]
```

ehmry, 2018-09-13:
@chelmuth fixup for that and libc_noux at 354a6b1b2ec685fe6c312b850e664a1f7327b6d9.

ehmry, 2018-09-13:
@chelmuth and an lwip fixup at 53d687786baa88d2f809d1717d7bd51ecddc0a0b.

chelmuth, 2019-04-09:
Another issue where I'm not sure about the state.

ehmry, 2019-04-10:
If we want to remove the `Insufficient_buffer` exception from the VFS, this is the issue to work from.

nfeske, 2021-02-26:
The topic has become stale.

commit 5d7b8321 (2018-09-13): Refactor VFS errors

Remove VFS errors not returned by any of the VFS plugins and replace the
"Insufficient_buffer" exception with "WRITE_BLOCKED" and
"WRITE_OVERSIZED". This generates a compile-time error when the VFS
write call is used directly but these two errors are not handled by a
switch statement. Retrying writes is not handled by the VFS, it is the
callers responsibility to segment buffers or dispatch I/O signals and
call write again.

The "Insufficient_buffer" exception originated at the File_system client
within the VFS and was thrown when there was not enough space in the
packet st [... cut]
commit 51f7cc19 (2018-09-21): Refactor VFS errors

Remove VFS errors not returned by any of the VFS plugins and replace the
"Insufficient_buffer" exception with "WRITE_BLOCKED". This generates a
compile-time error when a VFS write call is used directly but this
errors is not handled by a switch statement. Retrying writes is not
handled by the VFS, it is the callers responsibility to dispatch I/O
signals and call write again.

The "Insufficient_buffer" exception originated at the File_system client
within the VFS and was thrown when there was not enough space in the
packet stream buffer to fit an entire write operation. Thi [... cut]
closed 2021-02-26 by nfeske