genodelabs/genode #2956: intel_fb.run: add acpica

m-stein, 2018-08-30:


nfeske, 2018-08-31:
@m-stein you labeled this issue as bug. It would be nice to name the issue after the problem/symptom you observe and add a brief description.

m-stein, 2018-09-03:
This commit is a quickfix for the problem:
0cc2a4520d9ce8874066142ae1c6ac9561acfc34 intel_fb.run: add acpica and poll

However, with this commit it is still necessary to let the intel_fb poll as there is yet no communication between ACPICA and intel_fb about DS state changes. Furthermore, it is known that the test sometimes hangs for half a minute when the DS state changes but it returns to normal operation afterwards.

nfeske, 2021-02-26:
@m-stein is now the time to finally merge this commit? I'm uncertain.

m-stein, 2021-02-26:
I'm so much out of this topic now that I cannot say what are the implications of this commit. Unfortunately the affected run script doesn't seem to be tested regularly. Maybe @alex-ab can help out here?

alex-ab, 2021-02-26:
I also can't say anything about the implications. It is up to you @m-stein, who worked at the topic intensively.

m-stein, 2021-02-26:
> I also can't say anything about the implications.
Thanks for the quick feedback.

> who worked at the topic intensively.
Sorry, didn't mean to bother you.

@nfeske I think we shouldn't merge the workaround as it targets a use case that we don't test regularly while it adds overhead (polling) and strange behavior (freezes).

nfeske, 2021-02-26:
Ok, I will close this issue then.

commit 463bd481 (2020-09-23): intel_fb.run: add acpica and poll

This solves problems with external monitors at laptops with docking
stations. Normally, when a docking station is plugged in or out, one of
the display interfaces of the laptop has to be switched between the
external ports of the laptop itself and those of the docking station.
This may be the responsibility of the ACPI depending on the hardware. It
is necessary to let the Intel FB poll for changes as there is yet not
communication protocol between ACPICA and Intel FB.

Fixes #2956
closed 2021-02-26 by nfeske